realsenseai/realsense-ros #1080: Map not generated using opensourcetracking launch file

aswinkumar2019, 2020-02-13:
Hello guys,
I tried to create slam using opensourcetracking launch file,the map was not generated,only the point cloud at that instant was visible when decay time was 0 as default as shown in (https://github.com/aswinkumar2019/erros/raw/master/WhatsApp%20Image%202020-02-14%20at%201.44.46%20AM.jpeg) ,i changed the decay time to 100 and got some point cloud map like (https://github.com/aswinkumar2019/erros/raw/master/WhatsApp%20Image%202020-02-14%20at%201.44.46%20AM%20(1).jpeg) which is not actually good

What should be the decay time?
should i vary it or,i am doing something wrong?
Regards,
Aswin Kumar

aswinkumar2019, 2020-02-20:
Anybody please reply

doronhi, 2020-02-20:
Is it possible that you are viewing the wrong point cloud topic with RViz? Are you showing the output of realsense2_camera or of rtabmap?
I can't really tell what the problem is without more information.
It might help to see the command line and output of realsense2_camera and rtabmap from the terminal.

aswinkumar2019, 2020-02-20:
Thank you for your reply.
The pointcloud of realsense was subscribed,when i changed it to rtabmap,i dont get any output in rviz
The command line output is:
[ INFO] [1582205788.040527817]: SELECTED BASE:Depth, 0
[ INFO] [1582205788.135805495]: RealSense Node Is Up!
 20/02 19:06:28,167 WARNING [546836558208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/02 19:06:28,228 WARNING [546836558208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/02 19:06:28,355 WARNING [546836558208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/02 19:06:28,419 WARNING [546836558208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1582205788.544403005]: Odom: quality=0, std dev=99.995000m|99.995000rad, update time=0.148570s
[ WARN] (2020-02-20 19:06:28.813) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 19/20 (matches=21) between -1 and 2"
[ INFO] [1582205788.816553502]: Odom: quality=19, std dev=0.000000m|0.000000rad, update time=0.252049s
[ WARN] (2020-02-20 19:06:29.138) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 3"
[ INFO] [1582205789.144246794]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.311837s
[ INFO] [1582205789.396872546]: rtabmap (1): Rate=1.00s, Limit=0.000s, RTAB-Map=0.7144s, Maps update=0.0354s pub=0.0016s (local map=1, WM=1)
[ WARN] (2020-02-20 19:06:29.519) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=8) between -1 and 4"
[ INFO] [1582205789.524144091]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.375390s
[ WARN] [1582205789.805326833]: Transform from camera_imu_optical_frame to camera_link was unavailable for the time requested. Using latest instead.

[ WARN] (2020-02-20 19:06:29.843) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 5"
[ INFO] [1582205789.858323361]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.318506s
[ERROR] (2020-02-20 19:06:29.892) Rtabmap.cpp:1075::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 24 is ignored!
[ INFO] [1582205789.892473763]: rtabmap (1): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0169s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[ WARN] (2020-02-20 19:06:30.136) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 6"
[ INFO] [1582205790.147118680]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.285490s
[ WARN] (2020-02-20 19:06:30.426) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 16/20 (matches=23) between -1 and 7"
[ INFO] [1582205790.428815983]: Odom: quality=16, std dev=0.000000m|0.000000rad, update time=0.273477s
[ WARN] (2020-02-20 19:06:30.754) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=16) between -1 and 8"
[ INFO] [1582205790.759146221]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.272155s
[ WARN] (2020-02-20 19:06:31.047) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 9"
[ INFO] [1582205791.049987755]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.277200s
[ERROR] (2020-02-20 19:06:31.076) Rtabmap.cpp:1075::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 44 is ignored!
[ INFO] [1582205791.077205417]: rtabmap (1): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0005s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[ WARN] (2020-02-20 19:06:31.338) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=18) between -1 and 10"
[ INFO] [1582205791.346700345]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.287994s
[ WARN] (2020-02-20 19:06:31.627) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=13) between -1 and 11"
[ INFO] [1582205791.630701050]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.278674s
[ WARN] [1582205791.808400400]: Transform from camera_imu_optical_frame to camera_link was unavailable for the time requested. Using latest instead.

[ WARN] (2020-02-20 19:06:31.900) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 12"
[ INFO] [1582205791.908012618]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.268630s
[ERROR] (2020-02-20 19:06:31.931) Rtabmap.cpp:1075::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 63 is ignored!
[ INFO] [1582205791.932384930]: rtabmap (1): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0006s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[ WARN] (2020-02-20 19:06:32.238) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=6) between -1 and 13"
[ INFO] [1582205792.247689316]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.316322s
[... cut ...]
[ INFO] [1582205793.445697761]: Odom: quality=17, std dev=0.000000m|0.000000rad, update time=0.336962s
[ WARN] (2020-02-20 19:06:33.779) OdometryF2M.cpp:538::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=19) between -1 and 18"
[ INFO] [1582205793.789771916]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.329302s
[ WARN] [1582205793.809792576]: Transform from camera_imu_optical_frame to camera_link was unavailable for the time requested. Using latest instead.
Regards,
Aswin Kumar

doronhi, 2020-02-20:
When you switch to rtabmap's pointcloud it is possible that you also need to change the frame_id. This is from memory...
Also, the notifications about "Not enough inliers..." means that you should put the camera stable with an "interesting" view.
Also, the warning "control_transfer returned error..." is a realsense warning and an should be dealt with in a separate issue. See [here](https://github.com/IntelRealSense/realsense-ros/issues/1085#issuecomment-589020556) I don't know what it means yet but it could be problematic.

aswinkumar2019, 2020-02-20:
Thanks for the help,
i just adjusted the position of realsense and,just got a single warning:
[ WARN] [1582206950.053855198]: Transform from camera_imu_optical_frame to camera_link was unavailable for the time requested. Using latest instead.

And,the point cloud was generated successfully but,
after sometime,i get the Not enough inliers..."  error and,the point cloud stops.
Is there anything i can do to fix it?
Also,the point cloud was generated very slowly,is it the usual case or,i can increase the rate by modifying some parameters?
Regards,
Aswin Kumar

aswinkumar2019, 2020-02-20:
Also,
When you switch to rtabmap's pointcloud it is possible that you also need to change the frame_id.
By this line may i know which frame id you are referring to?
Regards,
Aswin Kumar

doronhi, 2020-02-23:
I just ran it now.
The topic of the point cloud is "/voxel_cloud"
The frame_id is "map"
It's not stable and you need to keep it looking at scenes with features but it reacts smoothly on my machine.

aswinkumar2019, 2020-02-23:
Hello doronhi,
What is the hardware you are using?(I am running it in jetson nano)
The default was voxel_cloud, I changed it to rtabmap_cloud.
Is it wrong?
Regards, 
Aswin Kumar

aswinkumar2019, 2020-02-23:
I didnt get map with voxel_cloud, so I thought it is point cloud from realsense and, changed it to rtabmap_cloud

doronhi, 2020-02-23:
There are known issues with the Jetson. Try to run the realsense2_camera node without the the rtabmap node and check that all the expected topics are stable. If not, please describe the behavior and errors you see. You should also check other open issues for issues regarding jetson, Raspberry pi, maybe they already described the problem you are facing.
It is possible that the issues are still being investigated.
If you can, I also suggest you try running on a different linux machine, to verify your configuration.

aswinkumar2019, 2020-02-23:
Will it run perfectly without any issues in upsquared?,if not, can you mention some hardwares in which it is officially tested? 
Is subscribing to rtabmap_cloud right or, I need to subscribe to voxel_cloud?
Regards, 
Aswin Kumar

doronhi, 2020-02-23:
voxel_cloud seems better and updates fluently.
I tried it a while ago on UP2 and it works but you'll have to set lower frame rates as this is not such a strong machine.

aswinkumar2019, 2020-02-23:
Thank you very much doronhi for your help. 
Just a small request,please keep the Intel SAWR project at github.com/intel/SAWR active.it is not updated for a long time and, is not monitored. 
It was of great help to start. 
Regards, 
Aswin Kumar

closed 2020-02-23 by aswinkumar2019

doronhi, 2020-02-24:
The SAWR project is indeed very old.
You should check out the [hedgehog](https://www.intelrealsense.com/hedgehog-robot-by-erwhi/). It is a maintained opensource project explaining about navigation, slam etc. with up to date hardware. It also directs you to it's [github repository](https://github.com/gbr1/erwhi-hedgehog).