realsenseai/realsense-ros #1083: Color image and PointClould synchronization

wikpunGitHub, 2020-02-15:
Hi, I use color image to filter out a moving object and I use PointCloud to determine it XYZ. However, I notice that if the object is moving fast, the XYZ position is not accurate. I would like to know how to synchronize the color image with the PointCloud. I am current setting my D435 in 60fps for both color and depth channel. 

I tried to use depth image before, the problem with depth image is that I can only get the depth  not the position in the other 2 axis. 

Thanks a lot for any help!

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

Hi @wikpunGitHub,

Can you please elaborate regarding the following:
what is the context of the question?
what are you trying to achieve (object removal, object recognition, object measurements, etc.) and what steps are you doing in order to achieve it?
what is your indication of "the XYZ position is not accurate"?
what is your system configuration(build number of RealSense SDK 2.0,build number of ROS Wrapper,etc.)?

Thank you!

wikpunGitHub, 2020-03-08:
Hi, I found out my issue is due to the fact 2 color images may come before the PointCloud. So I occasionally detect the fast moving image in the second color image while I am reading its coordinates in the first PointCloud. Thanks for the support!

closed 2020-03-08 by wikpunGitHub