realsenseai/realsense-ros #1085: Error extracting usb port from device with physical ID

anbello, 2020-02-16:
I am reporting this issue because I read this after launching realsense2_camera node:

[ WARN] [1581867815.081746419]: Error extracting usb port from device with physical ID: 1-1.3-6
Please report on github issue at https://github.com/IntelRealSense/realsense-ros

Follow complete output:

ubuntu@anbelloros:~$ roslaunch realsense2_camera rs_t265.launch
... logging to /home/ubuntu/.ros/log/6fd5b818-d0dc-11e5-b1e1-83e5777724cb/roslaunch-anbelloros-5447.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://anbelloros.local:35015/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 62
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: t265
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 800
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 848
 * /camera/realsense2_camera/gyro_fps: 200
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: camera/odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://anbelloros.local:11311

process[camera/realsense2_camera_manager-1]: started with pid [5464]
process[camera/realsense2_camera-2]: started with pid [5465]
[ INFO] [1581867814.514208834]: Initializing nodelet with 4 worker threads.
[ INFO] [1581867814.971967913]: RealSense ROS v2.2.12
[ INFO] [1581867814.972099627]: Running with LibRealSense v2.32.1
[ INFO] [1581867815.034546461]:  
[ INFO] [1581867815.078693388]: Device with serial number 0000925122110423 was found.

[ INFO] [1581867815.079251027]: Device with physical ID 1-1.3-6 was found.
[ INFO] [1581867815.079625442]: Device with name Intel RealSense T265 was found.
[ WARN] [1581867815.081746419]: Error extracting usb port from device with physical ID: 1-1.3-6
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[ INFO] [1581867815.111920128]: No calib_odom_file. No input odometry accepted.
[ INFO] [1581867815.112544224]: getParameters...
[ INFO] [1581867815.518296356]: setupDevice...
[ INFO] [1581867815.519296951]: JSON file is not provided
[ INFO] [1581867815.520063386]: ROS Node Namespace: camera
[ INFO] [1581867815.520826956]: Device Name: Intel RealSense T265
[ INFO] [1581867815.521634639]: Device Serial No: 0000925122110423
[ INFO] [1581867815.522459093]: Device physical port: 1-1.3-6
[ INFO] [1581867815.523189904]: Device FW version: 0.2.0.926
[ INFO] [1581867815.523912590]: Device Product ID: 0x0B37
[ INFO] [1581867815.524638974]: Enable PointCloud: Off
[ INFO] [1581867815.525436762]: Align Depth: Off
[ INFO] [1581867815.526162678]: Sync Mode: Off
[ INFO] [1581867815.527019735]: Device Sensors: 
[ INFO] [1581867815.527946737]: Tracking Module was found.
[ INFO] [1581867815.528804628]: (Depth, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1581867815.529571375]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1581867815.530292967]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1581867815.530982426]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1581867815.531815317]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1581867815.532672582]: num_filters: 0
[ INFO] [1581867815.533358394]: Setting Dynamic reconfig parameters.
[ INFO] [1581867815.595084677]: Done Setting Dynamic reconfig parameters.
[ INFO] [1581867815.597761731]: fisheye1 stream is enabled - width: 848, height: 800, fps: 30, Format: Y8
[ INFO] [1581867815.599591206]: fisheye2 stream is enabled - width: 848, height: 800, fps: 30, Format: Y8
[ INFO] [1581867815.600541229]: setupPublishers...
[ INFO] [1581867815.615667639]: Expected frequency for fisheye1 = 30.00000
[ INFO] [1581867815.773810339]: Expected frequency for fisheye2 = 30.00000
[ INFO] [1581867815.941032499]: setupStreams...
[ INFO] [1581867815.942220692]: insert Fisheye to Tracking Module
[ INFO] [1581867815.943048531]: insert Fisheye to Tracking Module
[ INFO] [1581867815.943670751]: insert Gyro to Tracking Module
[ INFO] [1581867815.944258286]: insert Accel to Tracking Module
[ INFO] [1581867815.944892172]: insert Pose to Tracking Module
 16/02 15:43:35,956 ERROR [1806255136] (tm-device.cpp:620) Streaming T265 video over USB 2.1 is unreliable, please use USB 3 or only stream poses
 16/02 15:43:35,956 ERROR [1806255136] (tm-device.cpp:620) Streaming T265 video over USB 2.1 is unreliable, please use USB 3 or only stream poses
 16/02 15:43:35,957 ERROR [1806255136] (tm-device.cpp:620) Streaming T265 video over USB 2.1 is unreliable, please use USB 3 or only stream poses
 16/02 15:43:35,957 ERROR [1806255136] (tm-device.cpp:620) Streaming T265 video over USB 2.1 is unreliable, please use USB 3 or only stream poses
[ INFO] [1581867815.984482989]: SELECTED BASE:Pose, 0
[ INFO] [1581867815.995168286]: RealSense Node Is Up!

doronhi, 2020-02-17:
Thank you. Could you please specify the machine type, and kernel version?

anbello, 2020-02-17:
Hi @doronhi I am using a Raspberry Pi 3B, now i don't remember the kernel version but I did an apt-get update - apt-get upgrade three days ago, anyway when I will be back to may home I will report the version.

anbello, 2020-02-18:
Here the kernel version on my RPi:
```
ubuntu@anbelloros:~$ uname -a
Linux anbelloros 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l armv7l armv7l GNU/Linux
```

doronhi, 2020-02-18:
Thanks. I have submitted PR#1088. I would appreciate it, if you have the time, to check it out and close the ticket if it works.

anbello, 2020-02-18:
Thanks. I will have the time to test it in the week end.

reedjacobp, 2020-02-20:
Hi @doronhi thank you for that fix. I have a D435i and I am using a Raspberry Pi 4 with the same kernel version as @anbello. I had the same error with extracting usb port and so I made the changes from the pull request and no longer get the error, but I am still receiving another error.

Please see the following:

```
ubuntu@BusBot:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/ubuntu/.ros/log/6e11a7b2-d0dc-11e5-8cfc-8fbcf91aa4e9/roslaunch-BusBot-8585.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://BusBot.local:37627/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 169 lines cut ...]
[ INFO] [1582169630.547272326]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1582169630.549396901]: setupPublishers...
[ INFO] [1582169630.567548158]: Expected frequency for depth = 30.00000
[ INFO] [1582169630.696581515]: Expected frequency for infra1 = 30.00000
[ INFO] [1582169630.812363396]: Expected frequency for infra2 = 30.00000
[ INFO] [1582169630.930427424]: Expected frequency for color = 30.00000
[ INFO] [1582169631.060535531]: setupStreams...
[ INFO] [1582169631.122665863]: insert Depth to Stereo Module
[ INFO] [1582169631.126631205]: insert Color to RGB Camera
[ INFO] [1582169631.127325428]: insert Infrared to Stereo Module
[ INFO] [1582169631.128573119]: insert Infrared to Stereo Module
[ INFO] [1582169631.129785032]: insert Gyro to Motion Module
[ INFO] [1582169631.131655132]: insert Accel to Motion Module
[ INFO] [1582169631.664612109]: SELECTED BASE:Depth, 0
 19/02 19:33:51,803 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1582169631.933717206]: RealSense Node Is Up!
 19/02 19:33:51,963 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,030 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,262 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...] messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,655 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,848 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

I'm not sure what's going wrong here. Please let me know if you need any more information.

doronhi, 2020-02-20:
@reedjacobp , I am glad the usb port issue is resolved.
I have encountered the other error you described and will update as soon as possible. Could you, as this is a different issue, open a new issue for it?

anbello, 2020-02-23:
@doronhi  I tested your PR #1088 and it solve the problem. Thanks.
One last thing I would like to know is why I see the lines with error for USB 2.1:
`16/02 15:43:35,956 ERROR [1806255136] (tm-device.cpp:620) Streaming T265 video over USB 2.1 is unreliable, please use USB 3 or only stream poses
`
even if I set "enable_fisheye1" and "enable_fisheye2" to false on launch file?

If I leave them to true I see 4 such error lines, if I set them to false I see 2 error lines.

doronhi, 2020-02-25:
The message that you see is just a message without any other effects.
As far as I understand it, the message is supposed to be issued only for enabling fisheye. Once disabling the fisheye cameras you can ignore it. To eliminate it completely you can also disable accel and gyro.
I issued a PR to fix it [here](https://github.com/IntelRealSense/librealsense/pull/5915)

closed 2020-02-25 by doronhi