realsenseai/realsense-ros #1086: ROS python: camera parameters

mmaceira, 2020-02-17:
Hello!

I've been working on a project where I detect objects in a RGB image. Now I project the detections to 3D with rs2_deproject_pixel_to_point as shown in:

https://github.com/IntelRealSense/librealsense/issues/1413

I have it working opening the camera with the realsense python wrapper ( example taken from https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py ).

Since I need the camera information for additional detections, I am planning to use ROS. I would like to read color and depth data with ROS. I run the following code to check the ROS installation:

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py

However, I'm not seeing any example on how to read the camera topic /camera/color/camera_info nor how to set up the pyrealsense2 from it. 

Alternatively, I thought of using the point cloud (with roslaunch realsense2_camera rs_camera.launch align_depth:=true enable_pointcloud:=true ), but if the point cloud is unorganized (https://github.com/IntelRealSense/realsense-ros/issues/851) I won't be able to get the 3D position from them.

There is some resource online that I'm missing? It would be very helpful some example showing how to obtain the 3D position of objects using the realsense-ros interface.

Thanks,
Marc

mmaceira, 2020-03-19:
Hello,

Any update / workaround for this?

Thanks,
Marc

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

@mmaceira Could you please check if http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29 on how to write a publisher and subscriber node in python helps you? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-04-17:
----------------------------------------------

@mmaceira Any other questions about this? Thanks!

mmaceira, 2020-04-17:
> @mmaceira Could you please check if http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29 on how to write a publisher and subscriber node in python helps you? Looking forward to your update. Thanks!

I have the publisher and subscriber node in python working properly. My problem is how to access  the realsense camera from ros.

I run this example from your repo without problems:
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py

However, I want to access the camera parameters to access the 3D position from the depth. I was looking for examples on reading the topics /camera/color/camera_info generated with the realsense.

Thanks for your time!

RealSenseCustomerSupport, 2020-06-16:
----------------------------------------------

@mmaceira Sorry that there's no such example in current SDK. Is this still problem for you? Looking forward to your update. Thanks!

mmaceira, 2020-06-17:
Yes, we are still blocked with it. It limits us since we don't know how (or if) it is possible to use the realsense and all the data it generates with ROS in a clean way.

RealSenseSupport, 2020-07-08:
@mmaceira Have submitted your requirement to our engineering team. Any update will let you know. Thanks!

doronhi, 2020-10-29:
Hi @mmaceira , Sorry for the extremely late response.
Will the example in the file "realsense2_camera/scripts/show_center_depth.py" in #1475 works for you?
It uses intrinsics from "/camera/depth/camera_info" topic to calculate 3D position of a pixel whos range is obtained from the depth image, received from the "/camera/depth/image_rect_raw" topic.

mmaceira, 2020-11-10:
Thanks for the update! I am working in a different project now, but a colleague is looking into it! I'll come back to you when we have checked it!

Thanks!

RealSenseSupport, 2020-11-18:
@mmaceira Any update from your side? Looking forward to your reply. Thanks!

acivit, 2020-11-20:
Hello! 

I'm @mmaceira's colleague, i checked if the example in #1475 worked properly, and it does. Since I'm interested in knowing the 3D position of detections from the color image, I'm using the "/camera/aligned_depth_to_color/image_raw" topic instead of "/camera/depth/image_rect_raw". 

Thanks!

doronhi, 2020-11-22:
Thanks for your confirmation. I think this issue can now be closed.

RealSenseSupport, 2020-11-23:
@acivit Thanks for your feedback! Close this issue accordingly.

closed 2020-11-23 by RealSenseSupport