realsenseai/realsense-ros #1087: Can't take the correct info from the topic "/camera/aligned_depth_to_color/image_raw" which is under rs_camera.launch

drfg0119, 2020-02-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR300 / D400 } | 
| Firmware Version                   |   | 
| Operating System & Version |   TX2(Ubuntu 18.04) & PC(Ubuntu 16.04)   | 
| ROS Version |   TX2(melodic) & PC(Kinetic) | 
| Kernel Version (Linux Only)    | 4.9                                      | 
| Platform                                 | NVIDIA Jetson TX2 (Jetpack4.2) |
| TX2 SDK Version                            |  {  2.31.0 & 2.22.0 }                          | 
| PC SDK Version                            |  {  2.31.0 & 2.24.0 & 2.32.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  Robot                          | 

### Issue Description
Hi, I got a problem about performance of SR305, when I run the API on TX2, the depth info was perfect, but if I run the node I wrote, there were incorrect values that only print 127mm at the maximum, and when I echo the topic, there was no value at the high byte position. But there was no problem when I run the node on PC.

Btw, all the parameters on SDK between PC and TX2 are the same, and the connection about D435i on TX2 and PC everything is good.

And these are the reference I followed to install on TX2:
SDK: [https://github.com/Tengyun-Mo/buildLibrealsense2TX2](url)
ROS wrapper: [https://github.com/jetsonhacks/installRealSenseROS](url)

Is there any people experienced this ..? Please help me.

It's the aligned depth to color image on PC.
[image]

And this is the incomplete frames on TX2.
[image]

But there is no problem when I run the API on TX2.
[image]

RealSenseCustomerSupport, 2020-03-24:
----------------------------------------------

Hi @drfg0119,

Can you please elaborate on what are you trying to achieve and what do you mean by "API on TX2",
"no value at the high byte position", etc?

Thank you!

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@drfg0119,

Do you still need help with this question?

drfg0119, 2020-04-03:
Oh hi @RealSenseCustomerSupport , 
Sorry, I had solved the problem, thak you!!!

closed 2020-04-03 by drfg0119