realsenseai/realsense-ros #1089: Control transfer returned error

reedjacobp, 2020-02-21:
Hi everyone,

So I had an issue with the "error extracting usb port" seen at #1085 but was able to fix that error. This time it says "RealSense Node is up!" and then I get:

WARNING [2856317984] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d

I can still see a depth cloud in RViz as well as a color image, but data is not being properly published to their topics and so they're not actually usable.

I am using a Raspberry Pi 4 running Ubuntu 16.04 with a D435i and kernel version 4.19.50.

Please see the following output:

```
ubuntu@BusBot:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/ubuntu/.ros/log/6e11a7b2-d0dc-11e5-8cfc-8fbcf91aa4e9/roslaunch-BusBot-8585.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://BusBot.local:37627/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 175 lines cut ...]
[ INFO] [1582169630.472113680]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1582169630.473676010]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1582169630.547272326]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1582169630.549396901]: setupPublishers...
[ INFO] [1582169630.567548158]: Expected frequency for depth = 30.00000
[ INFO] [1582169630.696581515]: Expected frequency for infra1 = 30.00000
[ INFO] [1582169630.812363396]: Expected frequency for infra2 = 30.00000
[ INFO] [1582169630.930427424]: Expected frequency for color = 30.00000
[ INFO] [1582169631.060535531]: setupStreams...
[ INFO] [1582169631.122665863]: insert Depth to Stereo Module
[ INFO] [1582169631.126631205]: insert Color to RGB Camera
[ INFO] [1582169631.127325428]: insert Infrared to Stereo Module
[ INFO] [1582169631.128573119]: insert Infrared to Stereo Module
[ INFO] [1582169631.129785032]: insert Gyro to Motion Module
[ INFO] [1582169631.131655132]: insert Accel to Motion Module
[ INFO] [1582169631.664612109]: SELECTED BASE:Depth, 0
 19/02 19:33:51,803 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1582169631.933717206]: RealSense Node Is Up!
 19/02 19:33:51,963 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,030 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,262 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,460 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,655 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 19:33:52,848 WARNING [2787111968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

doronhi, 2020-02-24:
@reedjacobp , I try to look into the case. What do you mean by "I can still see a depth cloud in RViz as well as a color image, but data is not being properly published to their topics and so they're not actually usable."?
Do you mean that the data is published at an uneven rate? Or that since the node crashes, the data is unreliable? or something else maybe?

reedjacobp, 2020-02-24:
@doronhi 
Thank you very much. What I mean by that is that when I pull up RViz, it shows a depth cloud in front of my robot model and there is an available live image if I add the "Image" visualization. Both of these are available even with the error above. I created my own package that utilizes rtabmap to build a map (which utilizes both of those topics), however, the map is not building. There could be something wrong with my package, but I believe it has something to do with the error.

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@reedjacobp Did you get the issue when you connect USB2 or USB3 port on your Raspberry Pi 4? If it's USB2, I would suggest you try to disable some streaming in the launch file(e.g. disable IR1 and IR2) to see if you still get the issue.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L21~#L22

doronhi, 2020-03-10:
The warning message "interrupt event received" can be ignored and indeed will be removed in future versions.
The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt. These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now.
I believe that if you can see the topics in RViz then as far as realsense2_camera node is concerned they are published correctly.

sadransh, 2020-03-11:
@doronhi as far as to test with these, it seems that this probably causes the point cloud ( or other depth/infrared topics) does not show up with "rostopic list". I did not experience the problem with similar versions of librealsense and realsense-ros on PC. I think the source of problem is related to libuvc builds on embedded devices(in my case jetson tx2)

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

@reedjacobp Any other questions with the reply from doronhi? Looking forward to your update. Thanks!
@sadransh What's your issue details? Could you please create another one to specify more? Thanks!

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

@reedjacobp Any update from your side? Thanks!

reedjacobp, 2020-03-27:
@doronhi @RealSenseCustomerSupport I apologize for the late reply, but I believe you are right about everything being published correctly. I can see the cloud in RViz, so it must work. I had something configured incorrectly so I wasn't visualizing it correctly in RViz. Thank you so much for your help!

closed 2020-03-27 by reedjacobp

sadransh, 2020-05-30:
@reedjacobp Did the Control transfer returned error disappeared for you? when using libuvc?