realsenseai/realsense-ros #1090: SLAM Error Realsense T265

jakelloyd22, 2020-02-25:
Hey guys,
after running the command:
roslaunch realsense2_camera rs_t265.launch
Rviz pulls up and I can see both image outputs from the T265's fish eye lenses; however, when I try to add outputs for other topics like the IMU and the odometry it doesn't work. Eventually the node fails and I get the following error:

 25/02 15:36:56,917 WARNING [139845745768192] (tm-device.cpp:1289) SLAM_ERROR Speed
 any one has an idea as to whats happening?

MartyG-RealSense, 2020-02-26:
I hope that the link below will provide useful information about accessing an odometry topic.  Please read downward through the comments from the point in the discussion that is linked to.

https://github.com/IntelRealSense/realsense-ros/issues/749#issuecomment-532968059

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @jakelloyd22

Is there any more questions or is the issue seen still after reviewing the information MartyG-RealSense provided?

**If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport

rakshith95, 2020-03-18:
> I hope that the link below will provide useful information about accessing an odometry topic. Please read downward through the comments from the point in the discussion that is linked to.
> 
> [#749 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/749#issuecomment-532968059)

Hello, I'm getting these warning/error messages as well when there is sharp movement to the T265. Not sure how the other thread addresses it

dwiel, 2021-07-20:
@RealSenseCustomerSupport I am having this error as well, especially with fast movement as @rakshith95 mentioned. I've read the linked thread [here](https://github.com/IntelRealSense/realsense-ros/issues/749#issuecomment-532968059), but it says nothing at all about this error message. Maybe you sent the wrong link?

MartyG-RealSense, 2021-07-20:
Hi @dwiel  I am not involved in support of the T265 Tracking Camera model.  Please create a new case on the front page of this forum that includes 'T265' in the message title by visiting the link below and left-clicking on the green **New Issue** button.

https://github.com/IntelRealSense/realsense-ros/issues