realsenseai/realsense-ros #1091: Change default axis of t265 to upward facing

2vin, 2020-02-27:
I want to put realsense t265 in upward facing configuration on a robot, but this gives me an orientation problem on the topic /camara/odom/sample in ROS where the odometry vector always face upwards. Is there a way to use the camera in upward facing configuration?

MartyG-RealSense, 2020-02-28:
A long discussion in the link below about using the T265 in downward orientation may provide advice that is adaptable for an upward orientation.

https://github.com/IntelRealSense/librealsense/issues/4080

dookei, 2020-03-30:
Hi @2vin ,
for our project we also need to use the T265 in upward configuration in order to "Track" the ceiling instead of ground-floor.Did you try this configuration any further?

2vin, 2020-03-30:
> Hi @2vin ,
> for our project we also need to use the T265 in upward configuration in order to "Track" the ceiling instead of ground-floor.Did you try this configuration any further?

Hi @dookei, Sorry unfortunately I could not find right information about the same. So I had to switch to forward facing configuration in ROS. Please let me know if you find any reliable solution.

dookei, 2020-04-01:
Hi @2vin looks like this was fixed 6 hours ago ;-). Just trying it now. Will keep you posted

[https://github.com/IntelRealSense/librealsense/pull/6176](url)

dookei, 2020-04-01:
Hi @2vin @MartyG-RealSense  so the firmware 0.2.0.951 fixed the issue like a charm- No more random yaw ;-)

MartyG-RealSense, 2020-04-01:
@dookei   Thanks so much for the update about your problem being resolved   :)

2vin, 2020-04-01:
> Hi @2vin @MartyG-RealSense so the firmware 0.2.0.951 fixed the issue like a charm- No more random yaw ;-)

Thanks for the update @dookei it should help a lot :)

closed 2020-09-10 by RealSenseSupport