realsenseai/realsense-ros #1092: T265 suddenly doesn't provide any new message at rostopic odometry

LEGO999, 2020-02-28:
Hi guys, I am using a raspberry pi 4, Ubuntu 18.04, D435i and T265 to build a indoor SLAM robot.
My SDK version is v2.32.1 and built using LibUVC backend. The firmware version for T265 is 0.2.0926.
I use rs_t265.launch to launch T265 and obtain my odometry of robot. My setup works for a few seconds and could build the 2D MAP. But after a few seconds, the T265 randomly doesn't provide new message in the rostopic /t265/odom/sample(t265 is the name I use) , the rostopic is still there but no new message. I am using the origin USB-C cable and USB3.0 port. Could anyone please help me?

tiralonghipol, 2020-03-27:
Hi @LEGO999 ,
I have the same problem! Were you able so solve the issue? 
I tried:
- same setup, another identical raspi 4 to check if I am really unlucky and i got a defective board
- power from stabilized device (5.1v, 3A) to check if it is a power-related problem
- adding an active cooling fan to check if it is heating-related 
- analyze /var/log/messages to check if there is any issue on usb
- switch from Ubuntu 18 to Raspian Buster to check if it is OS-related

I really dont know where to crash my head, any ideas?
I am 80% sure its a raspberry-side problem, because if you exclude ros the same problem happen in realsense-viewer

P.

LEGO999, 2020-03-27:
Hi @tiralonghipol ,

I agree with you very much, even when pure realsense-viewer is used, T265 will randomly crash. I think it could be the problem of the linux kernel. Currently, I am using 4.19.86-v8-25 on RPI4. And on my desktop with 5.3.0.40, such problem never happened(could be just lucky). Unfortunately, I am not an expert in this area and not able to fix this issue.

tiralonghipol, 2020-03-27:
I forgot to mention:
- increased usb buffer size from 16 to 20148 mb
- same test on Up Board -> no problem at all

Now its becoming a kind of obsession, I have to make it work !

RealSenseSupport, 2020-09-11:
Thank you for highlighting the issue while using T265. We have moved our focus to our next generation of products and consequently, if this issue is in T265, we will not be addressing this issue.

closed 2020-09-17 by RealSenseSupport