realsenseai/realsense-ros #1093: T265  odom in,ROS wrapper specific  questions

FPSychotic, 2020-02-28:
Hi, I cannot find answer to many basic qustions related with odometry in on T265.
1- SDK needs a configuration file (json), is needed for ROS wrapper? 
2- For every case? my wheel odom_link is on the TF tree already, and t265_link knows the TF already of wheel odom. in origen and position as t265----base_link----planar-odom.
3-If I need the config file, where I put the file
4- With what frequency t265 fuse the odometry in with its own?
5-How I know t265 is receiving the external odometry and processing it?
6-Im using this format, it is ok?
frame_id: "/my_odom"
child_frame_id: "/base_link"
pose: 
  pose: 
    position: 
      x: 0.0502667425837
      y: 0.0746460541491
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: -0.158547533432
      w: 0.987351345592
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: -3.32270851082e-05
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: -0.00683963986254
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
7-My combo d435 and t265 knows the wheel odom. TF, I dont use config file, but in RVIZ, the fish eye camera vieo with overlayed data,  the TF axes drift foward  with low light, but I see how every half or one second is corrected and comeback to the right position, I think it is because the odometry input is being fused properly, maybe because my TF tree and msg_odom are right and t265 dont need the config file,is it possible? Also I can see on the point cloud map, vertical drift is a lot bigger than planar drift, again a clue that odom in fusion is working. 
8- A topic telling if the odom. in is working and the topic name and rate should be great
Thanks

Thank,

sadransh, 2020-05-25:
@FPSychotic Did you find the answers?

FPSychotic, 2020-11-03:
I couldn't find

closed 2020-11-03 by FPSychotic