realsenseai/realsense-ros #1094: Realsense D400 with gazebo

q8wwe, 2020-02-29:
Can I use intel realsense SDK with gazebo .
I want to work on it with gazebo simulation before test on real robot .therefor ,  can compile this wrapper and use it with gazebo simulation ?

MartyG-RealSense, 2020-02-29:
There is a tutorial for using Gazebo for ROS that was designed for Kinect but says it can be used with other depth cameras.

http://gazebosim.org/tutorials?tut=ros_depth_camera&cat=connect_ros

There was an official RealSense Gazebo plugin a couple of years ago but it was designed for the older RealSense R200 stereo camera model.

https://github.com/intel/gazebo-realsense

q8wwe, 2020-02-29:
thank you , I will check it.

q8wwe, 2020-03-02:
I'm still looking for a stimulation I searched around but still can't make it work with gazebo .

MartyG-RealSense, 2020-03-02:
Recently there was an article on the RealSense blog about building your first RealSense robot, called the ERWHI Hedgehog.

https://www.intelrealsense.com/hedgehog-robot-by-erwhi/

The main website for the ERWHI Hedgehog robot project says that it can be simulated in Gazebo Simulator.

https://gbr1.github.io/erwhi_hedgehog.html

If you investigate the details of the ERWHI Hedgehog project's construction then this may provide one of the most user-friendly ways to get a modern RealSense camera working with a Gazebo simulation.

q8wwe, 2020-03-03:
Thank you , i will check and come back

closed 2020-05-30 by MartyG-RealSense