realsenseai/realsense-ros #1096: T265's IMU topic has quaternion of zeros

mhaboali, 2020-03-02:
I'm using demo_t265.launch with linter_interpolation filter to get /t265/imu topic. But the topic has a quaternion of zeros under the orientation section of the Imu message.

However, the orientation is correct from the /t265/odom/sample topic. But I need the orientation values to come directly from the /t265/imu topic.

Any help?

MartyG-RealSense, 2020-03-02:
I have seen similar cases about difficulty in publishing the T265's IMU data to a topic directly.  Please read down through the comments from the point in the conversation linked to below.

https://github.com/IntelRealSense/realsense-ros/issues/749#issuecomment-488748528

The first comment in this sequence provides a link highlighting where to map the odometry topic to.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/includes/nodelet.launch.xml#L74

mhaboali, 2020-03-05:
Thank you so much for the information @MartyG-RealSense 

I'm reading the details soon.

mhaboali, 2020-03-06:
@MartyG-RealSense 

I've just got a chance to look deeper into the comments and tried out most of them, but unfortunately, I didn't get the IMU's readings on the /t265/imu

Could you give an example or explain how I can get the IMU readings directly?

Your support is highly appreciated!

MartyG-RealSense, 2020-03-06:
I did further research.  On the T265, orientation data only is called 3-DOF, whilst orientation and position is called 6-DOF.  A T265 expert on the RealSense team said though that he has never seen 3-DOF (orientation only) occur during testing.

https://github.com/IntelRealSense/realsense-ros/issues/1055#issuecomment-590461015

If that is the case, perhaps it would be more realistic to publish 6-DOF to a topic if orientation-only mode is unlikely to be active.

mhaboali, 2020-05-11:
For the future related questions 

I Ended by using madgwick filter node to fuse the linear accelerations and angular velocities. And you can find opensource_tracking.launch file as an example.

Thanks,
Mohammed.

MartyG-RealSense, 2020-05-12:
Thanks so much for the update!   

Is it okay to close this case now if you have found a successful solution, please?   It will still be viewable by other people.

closed 2020-05-12 by mhaboali