realsenseai/realsense-ros #1098: rs_rgbd.launch launch file argument updates

swinterbotix, 2020-03-02:
Hi, 

I'm currently running the rgbd.launch file using a RealSense D435 camera. I noticed that the launch file does not include arguments 'enable_pose', 'enable_fisheye1', or 'enable_fisheye2' that can be set to false. As a result, I get the warning below...
[image]

The RealSense node still runs fine and realizes that it can't enable those streams, but it might be a good idea to add those arguments for robustness.

doronhi, 2020-03-08:
Hi,
realsense2_camera has many optional parameters, described in /launch/includes/nodelet.launch.xml.
The launch files, including rgbd.launch, were added as examples and easy starting points and weren't meant to include all the available options. One can always modify the launch files or create new ones.

The rgbd.launch file you mention really is not meant to be used with T265 so by default these values should really be set to false, as you suggest. We'd appreciate a pull request regarding it or we may take your suggestion and add those initial values to that file in the future.
Thank you for your attention.

P.S.
As you already notice, the node knows what sensors it can operate and the lack of setting those sensors to disable mode by the file changes nothing in its behavior.

swinterbotix, 2020-03-08:
Alright. Thanks for your response. Time permitting, I'll create a PR for this soon.

closed 2020-03-08 by swinterbotix

swinterbotix, 2020-08-12:
For anyone reading this thread, the defaults for the pose and fisheye arguments seemed to have been changed to 'false' in the nodelet launch file in tag 2.2.14!

doronhi, 2020-08-12:
> For anyone reading this thread, the defaults for the pose and fisheye arguments seemed to have been changed to 'false' in the nodelet launch file in tag 2.2.14!

It's been changed in version 2.2.15, as mentioned in the release notes: https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15

swinterbotix, 2020-08-12:
Oh yes, you're right. I meant to say 2.2.15.