realsenseai/realsense-ros #1104: running with docker

waspinator, 2020-03-04:
I'm trying to run ros within docker containers, but I'm having issue passing through the realsense hardware paths because they contain colons. Can you reconfigure the device names to not contain colons?

```yaml
version: "3"

services:

  ros-master:
    image: ros:kinetic-ros-core
    command: stdbuf -o L roscore
    networks:
      - ros
    restart: always

  realsense:
    build:
      context: ./docker
      dockerfile: ros-realsense.dockerfile
    environment:
      - "ROS_MASTER_URI=http://ros-master:11311"
      - "ROS_HOSTNAME=realsense"
    devices:
      - "/dev/HID-SENSOR-2000e1.3.auto:/dev/HID-SENSOR-2000e1.3.auto"
      - "/dev/iio:device0:/dev/iio:device0"
      - "/dev/iio:device1:/dev/iio:device1"
      - "/dev/video0:/dev/video0"
      - "/dev/video1:/dev/video1"
      - "/dev/video2:/dev/video2"
      - "/dev/video3:/dev/video3"
      - "/dev/video4:/dev/video4"
      - "/dev/video5:/dev/video5"
    networks:
      - ros
    restart: always
    command: roslaunch realsense2_camera rs_camera.launch

networks:
  ros:
    driver: bridge
```

waspinator, 2020-03-05:
adding the `privileged: true` option and sharing all of `/dev` seems to be working...

```yaml
version: "3"

services:

  ros-master:
    image: ros:kinetic-ros-core
    command: stdbuf -o L roscore
    networks:
      - ros
    restart: always

  realsense:
    privileged: true
    build:
      context: ./docker
      dockerfile: ros-realsense.dockerfile
    environment:
      - "ROS_MASTER_URI=http://ros-master:11311"
      - "ROS_HOSTNAME=realsense"
    devices:
      - "/dev:/dev"
    networks:
      - ros
    restart: always
    command: roslaunch realsense2_camera rs_camera.launch

networks:
  ros:
    driver: bridge
```

130s, 2020-11-05:
I concur the finding in https://github.com/IntelRealSense/realsense-ros/issues/1104#issuecomment-594974217. At my work we've been using Realsense driver (both librealsense and realsense2_camera) running within Docker container for more than a year with that.

130s, 2020-11-05:
Also related (and more active) discussion https://github.com/IntelRealSense/librealsense/issues/3849

RealSenseSupport, 2021-01-15:
Hi,

Will you be needing further help with this? Was @130s suggestion helpful?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

closed 2021-02-15 by RealSenseSupport

ll-nick, 2022-04-21:
So just an FYI for future travelers:
I didn't like mounting the entire `/dev` directory and giving the container privileged access. However, there I couldn't find any other way of doing online.

The following solution also works, although I have to use some bash magic, so I run my docker image using a `run.sh` script. Here's the TL;DR version, for more details see below:

## TL;DR

```bash
#!/bin/bash

BUS=$(lsusb | grep 8086 | cut -d " " -f 2)
PORT=$(lsusb | grep 8086 | cut -d " " -f 4 | cut -d ":" -f 1)

if [ ! -z "$PORT" ];
docker run \
  --volume /dev/bus/usb/$BUS/$PORT:/dev/bus/usb/$BUS/$PORT \
  --device-cgroup-rule "c 189:* rmw" \
  <a bunch more flags> \
  <img_name>
```

## Explanation

Instead of mounting the entire `/dev`, I only mount the specific device. Finding that wasn't super easy but using the bus and port info I get from `lsusb`, I can do that. E.g. `lsusb` delivers
```bash
Bus 004 Device 002: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i
```
You'll find the realsense at `/dev/bus/usb/004/002` with `004` being the first number of above output and `002` the second number.

To gain read/write access, instead of using the `--privileged` flag, I use `--device-cgroup-rule "c 189:* rmw"` where 189 is the major number of the USB bus. For more details see e.g. [this guide](https://www.matrix-vision.com/manuals/mvBlueFOX3/UseCases_section_mvBF3_in_docker.html).

solosuper, 2023-03-30:
> ... where 189 is the major number of the USB bus. 

I knew nothing about this magic number 189, so for other future travelers like me:

`ls -l /dev/bus/usb/${BUS}/${PORT}` shows something like
```
crw-rw-r-- 1 root root 189, 256 march  30 12:18 /dev/bus/usb/004/002
```
and there is the `189` again. The `c` in the beginning of the line is telling us the file type, where `file /dev/bus/usb/${BUS}/${PORT}`
 is even more explicit
```
/dev/bus/usb/004/002: character special (189/256)
```
[Googling "file character special"](https://www.ibm.com/docs/en/zos/2.3.0?topic=csf-character-special-files) tells me (emphasis mine) that
> A character special file is a file that provides access to an input/output device. Examples of character special files are: a terminal file, a NULL file, a file descriptor file, or a system console file. **Each character special file has a device major number**, which identifies the device type, and a device minor number, which identifies a specific device of a given device type.

That was all I could find out about **189**, and I'm kind of satisfied :) If someone has links to more complete information, please feel free to enlighten me (or us).

ciandonovan, 2023-06-10:
> I'm trying to run ros within docker containers, but I'm having issue passing through the realsense hardware paths because they contain colons. Can you reconfigure the device names to not contain [...]

I've submitted a pull request to the upsteam librealsense that should allow your original docker-compose to work as-is, without requiring the `--privileged` flag or needing the `--device-cgroup-rule "c 189:* rmw"` flag since that gives access to the entire USB which isn't ideal.