realsenseai/realsense-ros #1105: Setting up on Raspberry Pi

courtneyconroy, 2020-03-05:
Hi,

I am having some difficulty setting up ROS on the raspberry pi when following the instructions. It says to install ROS kinetic first and then clone in realsense-ros into the source folder. Which package do I need to install? Is it desktop_full? Do I do the complete setup for ROS kinetic and then add it into source?  I want to use this for occupancy mapping with the T265 and D435.

Thanks.

aswinkumar2019, 2020-03-05:
What is the problem you are facing?
Problem in installing realsense ros or, don't know the installation procedure?
If you have some problem in installing realsense ros package, give the problem you face while installing it. 
Installation procedure:
First, you have to install ros(install ros melodic)
Then, install librealsense, I think you have to install from source for raspberry pi(I am not sure about it).
Then, you have to install realsense ros package. 
Regards, 
Aswin Kumar

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@courtneyconroy For ROS Kinetic installation, please refer to http://wiki.ros.org/kinetic/Installation/Ubuntu

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@courtneyconroy Did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-04-01:
----------------------------------------------

@courtneyconroy  Could you please update? Thanks!

courtneyconroy, 2020-04-01:
Hi,

I decided to use Ubuntu mate 18.04 with ROS MELODIC. Worked fine then

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@courtneyconroy Thanks for the update! Close this accordingly.

closed 2020-04-08 by RealSenseCustomerSupport