realsenseai/realsense-ros #1106: realsensel D435/D453i  can not work

callmeSir007, 2020-03-06:
OS: UBUNTU 18.04 lts

it work well，when i use the version ：
SDK: 2.26.0
realsense-ros： 2.2.7
dmks ： 1.3.6
fw：5.11.6.250

now，upgrade the version to：
SDK: 2.31.0
realsense-ros： 2.2.12
dmks ： 1.3.13
fw：5.11.15.0/  5.12.1.0 / 5.12.1.100

it can not work，

v:/opt/ros/melodic$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/igv/.ros/log/515de9d2-5f44-11ea-9f16-54b2038b5c5e/roslaunch-igv-693.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://igv:42841/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
[... 81 lines cut ...]
[ INFO] [1583455768.601865945]: Done Setting Dynamic reconfig parameters.
[ INFO] [1583455768.660803074]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1583455768.661068407]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1583455768.661249961]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1583455768.706926867]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1583455768.707071436]: setupPublishers...
[ INFO] [1583455768.711322332]: Expected frequency for depth = 30.00000
[ INFO] [1583455768.755255756]: Expected frequency for infra1 = 30.00000
[ INFO] [1583455768.799320962]: Expected frequency for infra2 = 30.00000
[ INFO] [1583455768.861429294]: Expected frequency for color = 30.00000
[ INFO] [1583455768.910047237]: setupStreams...
[ INFO] [1583455769.018791394]: insert Depth to Stereo Module
[ INFO] [1583455769.018849716]: insert Color to RGB Camera
[ INFO] [1583455769.018870029]: insert Infrared to Stereo Module
[ INFO] [1583455769.018884937]: insert Infrared to Stereo Module
 06/03 08:49:29,109 WARNING [139700580902656] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ INFO] [1583455769.328613975]: SELECTED BASE:Depth, 0
[ INFO] [1583455769.375154188]: RealSense Node Is Up!
 06/03 08:49:29,385 WARNING [139700580902656] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 06/03 08:49:29,446 WARNING [139700580902656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/03 08:49:29,555 WARNING [139700580902656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/03 08:49:29,658 WARNING [139700580902656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/03 08:49:29,846 WARNING [139700580902656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/03 08:49:30,029 WARNING [139700580902656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
calR

jorgemia, 2020-03-06:
I get the same warning messages when trying to run on an Up Squared board with Ubuntu 16.04, Librealsense 2.33.1, ROS Kinetic, ros-realsense 2.2.12 and the D435i

doronhi, 2020-03-09:
@callmeSir007 , Thanks for the log file. I can see warning that are known and being investigated. Besides the uncomfortable warning messages, is there any topic that doesn't work or other behavior that prevent you from working?
@jorgemia ?
Thanks

jorgemia, 2020-03-09:
@callmeSir007 @doronhi I managed to fix these by applying PR #1088 (that didn't fix it immediately) and then running sudo apt purge ros-kinetic-librealsense2 as suggested in #1051 . After this, I ran catkin_make and no longer get issues when setting up the ros node.

callmeSir007, 2020-03-09:
@jorgemia @doronhi 

thank you for your response

I use the patched kernel modules instead of libuvc and now i can not see the warning message。

reference：
https://github.com/IntelRealSense/realsense-ros/issues/1073
https://github.com/IntelRealSense/librealsense/issues/5212

-------

Now I have another problem

i keep the camera running and display pointcloud data through rviz

I always get an error message：

1583734285.459592471 ERROR /camera/realsense2_camera_manager [/igv_ros_ws/src/sensor/realsense/realsense2_camera/src/base_realsense_node.cpp:1604(BaseRealSenseNode::frame_callback)] [topics: ] An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info

After a while，The process encounters an error and restarts automatically

Refer to the attachment for details of the log
[rosout.log](https://github.com/IntelRealSense/realsense-ros/files/4305743/rosout.log)

Why did this error occur？？

closed 2020-12-15 by RealSenseSupport