realsenseai/realsense-ros #1107: Error extracting usb port from device with physical ID: 2-1-3

jion8199, 2020-03-06:
My system is Ubuntu18.04
Follow the instructions to complete the installation
[image]
[image]
[image]

jion8199, 2020-03-06:
Added that my system is Jetson TX2, Ubuntu 18.4, ROS melodic,

RealSenseCustomerSupport, 2020-03-16:
----------------------------------------------

Hi @jion8199,

The issue was addressed in this PR: https://github.com/IntelRealSense/realsense-ros/pull/1088
Please let us know, if you need any further assistance with this matter.

Thank you!

closed 2020-03-31 by RealSenseCustomerSupport