realsenseai/realsense-ros #1108: Is it possible to change the pointcloud filter frequency rate?

schizzz8, 2020-03-06:
Hi,
My question is very simple. Looking at the `rs_camera.launch` present in the package it seems that there is no parameter to set the `pointcloud` filter publishing frequency. What I'd like to do is to keep the depth publishing at 30fps while reducing the publishing rate of the point cloud (also at 3-4Hz would be fine). Is it possible?

doronhi, 2020-03-09:
The simple answer is no. There is no parameter to set the pointcloud publishing rate.
There may be a workaround. The pointcloud filter publishes when both depth and it's texture are available. If you set color_fps:=6 you could still have the depth at 30Hz and the pointcloud will be published at 6Hz only.

schizzz8, 2020-03-09:
Thanks for the reply :)

closed 2020-03-09 by schizzz8