realsenseai/realsense-ros #1109: Details on enable_sync operation

araujokth, 2020-03-06:
Hi,

I was interested in getting for each IR frame the closest color frame, optimally if they would actually be triggered simultaneously. I noticed that there exists `enable_sync` which "gathers closest frames of different sensors, infra red, color and depth, to be sent with the same timetag. This happens automatically when such filters as pointcloud are enabled." 

However, I was wondering if someone could explain how this is exactly done, in particular how the timestamps are defined? Is it that the timetag of both IR and color images becomes the timestamp at which the IR image was triggered to be captured, OR when the color image was triggered to be captured, OR is the timestamp when the color image is done being captured, OR something else? @doronhi I saw your answer to this issue https://github.com/IntelRealSense/realsense-ros/issues/824#issuecomment-505759929 but I could not understand this aspect.

Another related question. For the timestamps of both IR images and color images with enable_sync=false, the timestamp always corresponds to the triggering time for capturing each of the images right?

Thanks again for the help!

doronhi, 2020-03-09:
When `enable_sync` is on, the frames are grouped together according to their capture time which, originated from the same device, is synchronized.
The timestamp given to the group by realsense2_camera is the ROS time at the publishing moment.

When enable_sync==false, the timestamps indeed corresponds to the capturing time of each image.

araujokth, 2020-03-12:
Sorry for the late reply and thanks a lot for the explanation!

closed 2020-03-12 by araujokth