realsenseai/realsense-ros #1110: fatal: You need to specify a tag name.

q8wwe, 2020-03-07:
Jeston TX2 J3.3.2 
Kernel : 4.4.159-tegra

while following the steps on installing librealsense ,my kernel didn't build after comand :  (./scripts/patch-realsense-ubuntu-lts.sh )

`Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.06.95-9build1).
build-essential is already the newest version (12.1ubuntu2).
git is already the newest version (1:2.7.4-0ubuntu1.7).
linux-headers-generic is already the newest version (4.4.0.174.182).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-xenial-master folder

Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name.
`

doronhi, 2020-03-09:
You cannot install kernel patches on Jetson. In the Jetson installation guide:  https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md, it says to install librealsense2-utils and librealsense2-dev but not librealsense2-dkms. For the same reason, the script will not work there.
If you build from sources and want to have access to all the HID information you should build with the flag FORCE_RSUSB_BACKEND = true. If you install the debian packages you'll get a version built with that flag on.

q8wwe, 2020-03-09:
By "can't install kernel patches on jetson " does this include manual installation ?

if i change "FORCE_RSUSB_BACKEND = true." will the installation work for me , through manual installation ?

and if so where i can find (FORCE_RSUSB_BACKEND = true.) .

doronhi, 2020-03-09:
The NVidia's linux is built as one unit and so doesn't allow patches.
Manual installation should work for you whether you specify this flag or not. rsusb backend gives you HID support without the kernel patches so if you don't use any IMU you may not need it at all.
The flag is located in CMake/lrs-options.cmake. You specify it, just like any other build flag, while running cmake command. For example:
```
cd build
cmake ../ -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=true -DBUILD_WITH_CUDA=true
make -j4
sudo make install
```

q8wwe, 2020-03-09:
My friend can you tell me the steps to do for making this work , I just want to install libreasense with a version stable enough to install ROs_wrapper package .

For days I was digging in this issue and there is a lot of complexity , between building the Kerenl for the UVC  and then you do it through manual installation but it didn't work and the process take a lot of time .

doronhi, 2020-03-09:
I am sorry if I made it unclear and for the trouble you've been through.
I refer to and copy from the guide you know: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Make Ubuntu up to date (I believe you already did):
`sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade`

`mkdir projects && cd projects` (or wherever you want)

Clone librealsense github repository:
`git clone https://github.com/IntelRealSense/librealsense.git`

Prepare Linux Backend and the Dev. Environment:
`sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev`
`sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev`

Run Intel Realsense permissions script located from librealsense root directory:
`./scripts/setup_udev_rules.sh`

Building librealsense2 SDK:
`mkdir build && cd build`
`cmake ../ -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true -DBUILD_WITH_CUDA=true`
`sudo make uninstall && make clean && make && sudo make install`

In which stage did thing went wrong?

q8wwe, 2020-03-09:
I went wrong on this stage : 
./scripts/patch-realsense-ubuntu-lts.sh
it gave :-  fatal: You need to specify a tag name.
but you didn't mention this step so is it okay to skip it ?

and I forget to mention that in the manual procedure whem it say check the new UVCvideo driver by the command (sudo dmesg | tail -n 50) it show nothing.

so I think the line (./scripts/patch-realsense-ubuntu-lts.sh) is important for updating the kernel and include UVC driver .

q8wwe, 2020-03-09:
I followed the instruction you listed up and after I run :Realsesne_viewer

I got warning : 
 
` 09/03 15:56:35,093 WARNING [547474108880] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/03 15:56:35,280 WARNING [547474108880] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61`

doronhi, 2020-03-10:
The warning message "interrupt event received" can be ignored and indeed will be removed in future versions.
The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt. These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now.

The compilation error suggest you link with an older version of librealsense2. Is it possible that you have an older version installed? (maybe `apt list --installed | grep librealsense2` can help?)

q8wwe, 2020-03-10:
nvidia@tegra-ubuntu:~/catkin_ws$ catkin_make clean
Base path: /home/nvidia/catkin_ws
Source space: /home/nvidia/catkin_ws/src
Build space: /home/nvidia/catkin_ws/build
Devel space: /home/nvidia/catkin_ws/devel
Install space: /home/nvidia/catkin_ws/install
####
#### Running command: "cmake /home/nvidia/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install -G Unix Makefiles" in "/home/nvidia/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at realsense-ros/realsense2_camera/CMakeLists.txt:37 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.32.1".

  The following configuration files were considered but not accepted:

    /opt/ros/kinetic/lib/aarch64-linux-gnu/cmake/realsense2/realsense2Config.cmake, version: 2.31.0
    /usr/lib/aarch64-linux-gnu/cmake/realsense2/realsense2Config.cmake, version: 2.28.1

CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:39 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

  

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
nvidia@tegra-ubuntu:~/catkin_ws$ apt list --installed | grep librealsense2

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

librealsense2/xenial,now 2.28.1-5ubuntu5~1.gbpb2a8ce arm64 [installed,automatic]
librealsense2-dev/xenial,now 2.28.1-5ubuntu5~1.gbpb2a8ce arm64 [installed]
librealsense2-gl/xenial,now 2.28.1-5ubuntu5~1.gbpb2a8ce arm64 [installed,automatic]
librealsense2-udev-rules/xenial,now 2.28.1-5ubuntu5~1.gbpb2a8ce arm64 [installed,automatic]
librealsense2-utils/xenial,now 2.28.1-5ubuntu5~1.gbpb2a8ce arm64 [installed]
ros-kinetic-librealsense2/xenial,now 2.31.0-1xenial-20191214-072450+0000 arm64 [installed]
nvidia@tegr

doronhi, 2020-03-10:
It says you still has librealsense2 versions 2.28.1 and also version 2.31.0.
Since you built from source, you now need to remove the debian packages:
try : `sudo apt purge librealsense2`
and perhaps `sudo apt purge ros-kinetic-librealsense2` as well.
Verify that all the debian packages were removed and try building again.

q8wwe, 2020-03-10:
unbelievable it's so hard and has a lot of errors, you solve one problem just to jump in another on.
up to now I flashed my Jeston TX2 over 15 times trying different approaches but non-working as expected.
I tried  different versions of L4t, and every time I try to install libreasense through: sudo apt-get install librealsense2-utils , I get an old version 2.28.1 and  this seem not gonna be solved any time soon. so I have to remove it and install libreasense through : sudo apt-get install ros-kinetic-realsense-camera , which will install 2.31.0 as you saw in the previous message that's why I have two versions.
even if I remove 2.28.1 and keep 2.31.0 i will run into a compile error that you have 2.31.0 and 2.33.1. 

it's like there is no win solution at all, shall I give up on this camera or what? troubleshoot these issues is really time-consuming!!

Mr. doronhi , I will try with the solution you provided

q8wwe, 2020-03-10:
apt list --installed | grep librealsense2 :-

dia@tegra-ubuntu:~$ apt list --installed | grep librealsense2

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

librealsense2-udev-rules/xenial,now 2.28.1-5ubuntu5~1.gbpb2a8ce arm64 [installed,auto-removable]

-------------------------------------------------------------------------
and it didn't work asking for V2.33.1

`Make Warning at realsense-ros/realsense2_camera/CMakeLists.txt:37 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2"
  (requested version 2.32.1) with any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:39 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

  

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
`

q8wwe, 2020-03-10:
even if I want to downgrade Ros_Wrapper there is no single release supporting V2.28.1, whatever package I'm using I got compile error "You have this version and you need this version ".
https://github.com/IntelRealSense/realsense-ros/releases
Actually I need a stabled version with less errors , here I'm trying my best but non of the releases are working .

doronhi, 2020-03-10:
@q8wwe , I am so sorry for the frustration and difficulties you are going through. You are right that no package should be so hard to install. It is true that on Jetson machine it sometimes is more tricky but your case seems quite the extreme and I am sorry about it.
I don't know how come you get older debian versions and as you said, whatever the reason is, you need a solution. So we keep to building from source.

As far as I understand it, the situation on your machine is like so:
1. You no longer have any version of librealsense2 installed.
2. You download and built librealsense2 from code. 
Before we managed that there is no point trying to build realsense2_camera.

As for the built version, I assume that at the end of the build process you ran 'sudo make install'. To make sure, please open a new terminal and type `realsense-viewer`. If the command failed it means that `sudo make install` failed.
The version appears on top of the realsense-viewer window and it reads 2.33.1. If the version is different I want you to check the branch and repository you cloned from. Check the version as it appears in the code in the file: librealsense\include\librealsense2\rs.h. What are the values of RS2_API_MAJOR_VERSION, RS2_API_MINOR_VERSION and RS2_API_PATCH_VERSION?

q8wwe, 2020-03-13:
I rebuilt librealsense on jetson TX2 and installed Ros wrapper I got the below warnings/errors.

Note : I tested the same on normal PC (desktop dell optiplex i7) with No GPU , libreasense installed through debian and I didn't face the same warnings/errors I got on Jetson TX2.
However , I think libreasense should work better on Jetson TX2 with its dedicated navidia GPU m but unfortunately that wasn't the case .
when it come to Jetson TX2 I faced a lot of complexity while installing libreasense SDK and warnings/errors in performance.

please advice .
libreasense version : 
`nvidia@jetson-0324917251038:~$ apt list --installed | grep librealsense2

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-kinetic-librealsense2/xenial,now 2.31.0-1xenial-20191214-072450+0000 arm64 [installed]
`
realsense-viewer show : 2.33.1
 warnings from realsense -viewer:
` 13/03 15:46:59,996 WARNING [548201988560] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 0, 
 13/03 15:47:00,000 WARNING [548201988560] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 0, 
 13/03 15:47:09,601 WARNING [548179100112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61`

warning and errors from : roslaunch realsense2_camera rs_camera.launch filters:=pointcloud 
`[ WARN] [1584103870.532427600]: Hardware Notification:USB REC overflow,1.5841e+12,Error,Hardware Error
 13/03 15:51:10,557 WARNING [547356668304] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 23, 
 13/03 15:51:11,143 WARNING [547390222736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1584103871.553157555]: Hardware Notification:IR stream start failure,1.5841e+12,Error,Hardware Error`

doronhi, 2020-03-17:
First, you have an older installation, v2.31.0, of librealsense2. It doesn't seem to cause any harm but you better remove it: `sudo apt purge ros-kinetic-librealsense2`
The "interrupt event.." message is just a message and will be removed in the near future. The "control_transfer.." message, as long as it appears just a couple of times, is being investigated but not known to cause any issues.
The "IR stream start failure" message can cause problems. Does it happen every run?

The "Out of frame resources!" message and the "Exception...processing callback!" could be related to the "IR stream failure" but more often they are related to lack of system resources - the app can't keep up with the incoming flow of images. For that you can reduce frame rate and resolution.
Building with CUDA support should really help in that case on NVidia machines. When you say "but unfortunately that wasn't the case ." do you mean that building with CUDA failed or that you didn't see much improvement by using CUDA on the NVidia machine?

q8wwe, 2020-03-19:
As suggested I removed V2.31.0 (sudo apt purge ros-kinetic-librealsense2) and I I lowered th frames to 15 fps   :

messenger-libusb : is continous warning .
messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

for IR stream start failure : I lowered the frames and reset (initial_reset:=true
). I didn't see the warning .

question about "messenger-libusb" is it safe warning !!

doronhi, 2020-03-20:
If the message "messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11" is continuous, that is appears multiple times per second, then it is not good. It suggest a real communication problem between the code and the device and can have effects. Can you add a log showing that?

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

Hi @q8wwe,

Do you still need help with this question?

q8wwe, 2020-03-31:
by log do you mean terminal info flow ?

`nvidia@jetson-0324917251038:~$ roslaunch realsense2_camera rs_d435_camera_with_model.launch 
... logging to /home/nvidia/.ros/log/798beba2-7371-11ea-994b-00044ba7dc33/roslaunch-jetson-0324917251038-27717.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-0324917251038:39501/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 880 lines cut ...]
 31/03 20:04:02,122 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:02,256 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:02,256 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:02,455 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:02,456 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:02,589 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:02,589 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:02,789 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:02,789 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:02,922 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:02,922 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:03,123 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:03,123 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:03,256 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:03,256 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:03,456 WARNING [546584916368] (types.cpp:49) Out of frame resources!
[... cut ...]
 31/03 20:04:03,790 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 31/03 20:04:03,923 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:03,923 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
^C 31/03 20:04:04,123 WARNING [546584916368] (types.cpp:49) Out of frame resources!
 31/03 20:04:04,123 ERROR [546584916368] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!`

RealSenseCustomerSupport, 2020-04-17:
----------------------------------------------

@q8wwe,

Can you please share with us your launch file?

closed 2020-05-01 by RealSenseCustomerSupport