realsenseai/realsense-ros #1111: 3D scanning examples of T265 + D435i combo?

VernonPerry, 2020-03-08:
I am working on a 3D scanning project and ordered a T265 and T435i to use with the calibrated 3D mount. But I only found a single example related to ROS robot navigation; are there any 3D scanning packages that support this dual tracking + depth camera array configuration, to as the article says increase the accuracy of 3D scanning applications?

Thanks in advance

MartyG-RealSense, 2020-03-08:
I do not know of any pre-made commercial or open-source software packages that have the built-in capability to use the T265 and D435i together.

It sounds as though you may have already seen Intel's tutorial for using tracking and depth together with a mount.

https://github.com/IntelRealSense/librealsense/tree/master/examples/tracking-and-depth

Another user asked a couple of months ago on the Intel Support RealSense forum about using a D435i and T265 together for real-time 3D scanning.

Once you have generated a point cloud scan with both devices using a method such as that in the tracking and depth tutorial, the point cloud could be exported as a **.ply** format point cloud data file.  This ply file can then be converted by various 3D programs into a solid-mesh 3D model.

https://support.intelrealsense.com/hc/en-us/community/posts/360038406074-Open-source-software-for-realtime-3D-scanning-with-the-D435i-and-T265

VernonPerry, 2020-03-08:
Ooh good find, I hadn't found that tutorial yet, just the first writeup that had a couple of different use cases and the 3D STLs for the calibrated mounting bracket. Thank you.

So it sounds like all I need to do is then is integrate this tracking and depth process with both cameras as our 3D scanning process that results in PLY files that can then be edited and cleaned up with MeshLab etc?

MartyG-RealSense, 2020-03-08:
Yes, once you have a .ply file then you can use it in a range of free 3D modelling programs, including:

- MeshLab
- Blender
- Pyntcloud (Python-based point cloud processing and mesh conversion)

Top-end programs such as Maya should be able to import a ply by using a plugin for ply comptability.

If you want a process that can provide more detail than just the solid 3D mesh, software called Artec Studio can be used with another program called SOLIDWORKS via a plugin called Geomagic for SOLIDWORKS.  The Geomagic plugin enables you to take a 3D model and convert it into CAD details and perform operations such as checking for deviation by comparing the scan to a 3D model.

https://www.artec3d.com/3d-software/solidworks-cad

closed 2020-09-21 by RealSenseSupport