realsenseai/realsense-ros #1114: Jeston TX2 libreasense sdk v2.33.1

q8wwe, 2020-03-08:
I installed Librealsense SDK for jetson TX2 following the link : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

which installed Librealsense sdk V2.28.1 , when I start to install realsense ros by following step No.2 on the link : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

I run into compile error because of the version ,

how can I update to SDK 2.33.1??
or 
which realsens-ros package is compatible with sdk v 2.28.1??

kindly help in solving this issue I been stumbling for days jumping from one problem to another trying to make things work out .
its really overwhelming as there is no straight guide on how to make Jetson TX2 work with D435.

kindly advise.

doronhi, 2020-03-09:
Apologizes for the lack of references to Jetson in the ROS installation guide. I'll try to attend to it.
For your question, did you `apt-get update`? The 2.33.1 version is not yet published as debian for Jetson but you should get 2.32.1 which should suffice for building realsense2_camera version 2.2.13.

q8wwe, 2020-03-09:
Thank you fro the prompt reply. I surely did sudo ap-get update and upgrade .
I tried different procedure to install librealsesne and Ros wrapper  but non worked.As you can sees I tried : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md
and also I tried manual installation https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
but i got error https://github.com/IntelRealSense/realsense-ros/issues/1110 (fatal: You need to specify a tag name)
kindly point me on the right  procedure ,how can I install libreasesne 2.33.1 or 2.32.1 ?

doronhi, 2020-03-09:
Regarding #1110, see my reply there. It involves the kernel patches but it's a different issue from building librealsense2. When you `apt-get install librealsense2-dev`, what version do you get?

q8wwe, 2020-03-09:
version 2.28.1

RealSenseCustomerSupport, 2020-03-18:
----------------------------------------------

Hi @q8wwe,

Can you please copy to this thread the compilation error you are talking about?

Thank you!

q8wwe, 2020-03-19:
Will its much easier to use Debian packages for installing libreasesne by using just the below instructions : 
1. 
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE

2. for ubuntu 16. : 
sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u

3. 
sudo apt-get install librealsense2-utils
sudo apt-get install librealsense2-dev

but the previous approach will install an old version, libreasense V2.28.1 which is not compatible with the new installation for the recent Ros wrapper (build 2.2.13) which fixed a lot of problems in the previous releases (example :build 2.2.12 and older ).

Note : I tried relsense V2.28.1 with different ROs wrsapper releases (https://github.com/IntelRealSense/realsense-ros/releases) and no single one is compatible  with v2.28.1.

the compatible problem if I remeber was like this : you have libreasense V2.28.1 and this version is not compitable with the installed ROS wrapper .

I'm sorry for my bad English if this isn't clear please let me know and I will try to explain it in better way.

Thank you for the effort and time you spend trying to help me.

Best regards

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

@q8wwe,

It would help us if you provide us with the details about compilation error you have mentioned in the issue description.

closed 2020-04-23 by RealSenseCustomerSupport