realsenseai/realsense-ros #1117: T265 No RealSense devices were found! after restart

droter, 2020-03-09:
Hi,

I am running a t265 on a Jetson Nano.  When I reboot the computer the node will not start.

[ INFO] [1583771648.864206033]: RealSense ROS v2.2.12
[ INFO] [1583771648.864318275]: Running with LibRealSense v2.31.0
[ WARN] [1583771649.406488769]: **No RealSense devices were found!**

$ lsusb
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
**Bus 001 Device 007: ID 03e7:2150**  

The camera is listed but will not show up in realsense-viewer unless I unplug it and plug it back in.

Any ideas?

Matt

MartyG-RealSense, 2020-03-10:
I have referred your case to a specialist T265 support engineer.  Thanks!

droter, 2020-03-12:
Looks like this is a known issue:
https://github.com/IntelRealSense/librealsense/issues/5913

closed 2020-03-12 by droter

AndreV84, 2021-01-28:
@MartyG-RealSense 
is there any easy-to-implement hardware or software solution that will waive the necessity to use human to detach/ attach the t265 after PC reboot?
We are looking to do it automaticallt - to get recognized automatically after Ubunbtu computer reboot
could you suggest an easy solytion for that, please?

RealSenseSupport, 2021-01-28:
Hi @AndreV84 

There may not be an easy to implement hardware solution.  I've seen or heard users control the VBUS power signal from the USB cable/connector with an external power supply to be able to toggle the power to reset the T265.  Also maybe the hardware reset command might work unless that has already been tried or attempted to be used.

AndreV84, 2021-01-29:
@RealSenseSupport 
this hardware reset? or different?
https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line