realsenseai/realsense-ros #1118: Install Instructions:  git checkout 'git tag ...  did not match any file(s) known to git.

droter, 2020-03-09:
In the install process it lists:
git checkout 'git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1'

I get the following error message:
:~/catkin_ws/src/realsense-ros$ git checkout 'git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1'
error: pathspec 'git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1' did not match any file(s) known to git.

doronhi, 2020-03-10:
You copied the upper comma wrong. It should be "`" and not "'".
That is:
```
git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`
```
and **NOT:**
```
git checkout 'git tag | sort -V | grep -P "^\d+.\d+.\d+" | tail -1'
```

droter, 2020-03-10:
Good catch.  

Thank you.

closed 2020-03-10 by droter