realsenseai/realsense-ros #1119: error: 'color_sensor' is not a member of ' rs2'

suk8267, 2020-03-10:
my OS : Ubuntu 16.04 LTS

after all this instruction,
i tried catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release

but it got an error like this

/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:704:32: error: ‘color_sensor’ is not a member of ‘rs2’
             else if (sensor.is<rs2::color_sensor>())
                                ^
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:704:32: error: ‘color_sensor’ is not a member of ‘rs2’
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:704:51: error: no matching function for call to ‘rs2::sensor::is()’
             else if (sensor.is<rs2::color_sensor>())
                                                   ^
In file included from /usr/local/include/librealsense2/hpp/rs_device.hpp:8:0,
                 from /usr/local/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/local/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/local/include/librealsense2/rs.hpp:9,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:11,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/local/include/librealsense2/hpp/rs_sensor.hpp:301:14: note: candidate: template<class T> bool rs2::sensor::is() const
         bool is() const
              ^
/usr/local/include/librealsense2/hpp/rs_sensor.hpp:301:14: note:   template argument deduction/substitution failed:
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:704:51: error: template argument 1 is invalid
             else if (sensor.is<rs2::color_sensor>())
                                                   ^
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:709:32: error: ‘fisheye_sensor’ is not a member of ‘rs2’
             else if (sensor.is<rs2::fisheye_sensor>())
                                ^
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:709:32: error: ‘fisheye_sensor’ is not a member of ‘rs2’
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:709:53: error: no matching function for call to ‘rs2::sensor::is()’
             else if (sensor.is<rs2::fisheye_sensor>())
                                                     ^
In file included from /usr/local/include/librealsense2/hpp/rs_device.hpp:8:0,
                 from /usr/local/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/local/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/local/include/librealsense2/rs.hpp:9,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:11,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/local/include/librealsense2/hpp/rs_sensor.hpp:301:14: note: candidate: template<class T> bool rs2::sensor::is() const
         bool is() const
              ^
/usr/local/include/librealsense2/hpp/rs_sensor.hpp:301:14: note:   template argument deduction/substitution failed:
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:709:53: error: template argument 1 is invalid
             else if (sensor.is<rs2::fisheye_sensor>())
                                                     ^
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:714:32: error: ‘motion_sensor’ is not a member of ‘rs2’
             else if (sensor.is<rs2::motion_sensor>())
                                ^
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:714:32: error: ‘motion_sensor’ is not a member of ‘rs2’
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:714:52: error: no matching function for call to ‘rs2::sensor::is()’
             else if (sensor.is<rs2::motion_sensor>())
                                                    ^
In file included from /usr/local/include/librealsense2/hpp/rs_device.hpp:8:0,
                 from /usr/local/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/local/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/local/include/librealsense2/rs.hpp:9,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:11,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/local/include/librealsense2/hpp/rs_sensor.hpp:301:14: note: candidate: template<class T> bool rs2::sensor::is() const
         bool is() const
              ^
/usr/local/include/librealsense2/hpp/rs_sensor.hpp:301:14: note:   template argument deduction/substitution failed:
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:714:52: error: template argument 1 is invalid
             else if (sensor.is<rs2::motion_sensor>())
                                                    ^
realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1162: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

please help me

doronhi, 2020-03-10:
It seems from the error messages you get that you have an older version of librealsense2 installed.
You need at least librealsense2 version v2.32.1.

droter, 2020-03-10:
I am getting the same build error.

These are the installed versions of librealsense2
librealsense2-dev is already the newest version (2.32.1-5ubuntu5~1.gbp83f952).
librealsense2-utils is already the newest version (2.32.1-5ubuntu5~1.gbp83f952).

suk8267, 2020-03-11:
i checked version of librealsense2 is that
apt list librealsense2
Listing... Done
librealsense2/xenial,now 2.33.1-0~realsense0.2043 amd64 [installed]
N: There are 85 additional versions. Please use the '-a' switch to see them

it seems v2.33.1-0

doronhi, 2020-03-11:
The functions that the build process can't find, i.e. sensor.is<rs2::color_sensor>() were added to librealsense starting from version v2.32.1 so they exist in v2.33.1-0.
Could it be that you have more then just one version installed? An older version in addition to v2.33.1 that you found?
When you open a new terminal and run `realsense-viewer`, what version is written at the window's top?

droter, 2020-03-12:
looks like the latest release on bionic is 2.32.1   

$ apt list librealsense2 -a
Listing... Done
librealsense2/bionic,now 2.32.1-5ubuntu5~1.gbp83f952 arm64 [installed,automatic]
librealsense2/bionic 2.31.0-5ubuntu5~1.gbp95d195 arm64
librealsense2/bionic 2.30.0-5ubuntu5~1.gbp9f99fa arm64

suk8267, 2020-03-13:
@doronhi i don't know about one version installed but i just install by apt-get install, so it may be no two version.

suk8267, 2020-03-13:
@doronhi and i checked version of realsense-viewer, it seems v2.33.1

suk8267, 2020-03-14:
@doronhi 
and this is cmake file

cmake_minimum_required(VERSION 2.8.3)
project(realsense2_camera)
add_compile_options(-std=c++11)

option(BUILD_WITH_OPENMP "Use OpenMP" OFF)
option(SET_USER_BREAK_AT_STARTUP "Set user wait point in startup (for debug)" OFF)

find_package(catkin REQUIRED COMPONENTS
    message_generation
    nav_msgs
    roscpp
    sensor_msgs
    std_msgs
    nodelet
    cv_bridge
    image_transport
    tf
    ddynamic_reconfigure
    diagnostic_updater
    )

if(BUILD_WITH_OPENMP)
    find_package(OpenMP)
    if(NOT OpenMP_FOUND)
        message(FATAL_ERROR "\n\n OpenMP is missing!\n\n")
    else()
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS} -fopenmp")
        set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} ${OpenMP_EXE_LINKER_FLAGS}")
    endif()
endif()

if(SET_USER_BREAK_AT_STARTUP)
	message("GOT FLAG IN CmakeLists.txt")
	set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DBPDEBUG")
endif()

find_package(realsense2 2.32.1)
if(NOT realsense2_FOUND)
    message(FATAL_ERROR "\n\n Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases\n\n")
endif()

if(NOT CMAKE_BUILD_TYPE)
    set(CMAKE_BUILD_TYPE Release)
endif()

string(TOUPPER "${CMAKE_BUILD_TYPE}" uppercase_CMAKE_BUILD_TYPE)
if (${uppercase_CMAKE_BUILD_TYPE} STREQUAL "RELEASE")
    message(STATUS "Create Release Build.")
    set(CMAKE_CXX_FLAGS "-O2 ${CMAKE_CXX_FLAGS}")
else()
    message(STATUS "Create Debug Build.")
endif()

set(CMAKE_CXX_FLAGS "-fPIE -fPIC -std=c++11 -D_FORTIFY_SOURCE=2 -fstack-protector -Wformat -Wformat-security -Wall ${CMAKE_CXX_FLAGS}")

add_message_files(
    FILES
    IMUInfo.msg
    Extrinsics.msg
    )

generate_messages(
    DEPENDENCIES
    sensor_msgs
    std_msgs
    )

set(CMAKE_NO_SYSTEM_FROM_IMPORTED true)
include_directories(
    include
    ${realsense2_INCLUDE_DIR}
    ${catkin_INCLUDE_DIRS}
    )

# RealSense ROS Node
catkin_package(
    LIBRARIES ${PROJECT_NAME}
    CATKIN_DEPENDS message_runtime roscpp sensor_msgs std_msgs
    nodelet
    cv_bridge
    image_transport
    ddynamic_reconfigure
    nav_msgs
    )

add_library(${PROJECT_NAME}
    include/constants.h
    include/realsense_node_factory.h
    include/base_realsense_node.h
    include/t265_realsense_node.h
    src/realsense_node_factory.cpp
    src/base_realsense_node.cpp
    src/t265_realsense_node.cpp
    )

add_dependencies(${PROJECT_NAME} ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME} ${catkin_EXPORTED_TARGETS})

target_include_directories(${PROJECT_NAME}
  PRIVATE ${realsense2_INCLUDE_DIR})

target_link_libraries(${PROJECT_NAME}
    ${realsense2_LIBRARY}
    ${catkin_LIBRARIES}
    ${CMAKE_THREAD_LIBS_INIT}
    )

# Install nodelet library
install(TARGETS ${PROJECT_NAME}
    ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
    LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
    RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
    )

# Install header files
install(DIRECTORY include/
    DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION}
    )

# Install launch files
install(DIRECTORY launch/
    DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/launch
    )

# Install rviz files
install(DIRECTORY rviz/
    DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/rviz
    )

# Install xml files
install(FILES nodelet_plugins.xml
    DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}
    )

guiolpei, 2020-03-14:
I had the same problem and fixed it compiling _librealsense_ from source instead of installing it from the repository.

Ubuntu 16.04.6 LTS with ROS kinetic.

droter, 2020-03-14:
Just finished installing librealsense from source and was able to build the ros package.

***
If you happen to be on a Jetson Nano using the installLibrealsense script from Jetson Hacks.  It doesn't re-download the latest git repository if it finds the directory.
https://github.com/JetsonHacksNano/installLibrealsense/issues/28

suk8267, 2020-03-16:
@guiolpei,@droter you means that i have to cone librealsense and just compile? and i don't know about where is the folder

suk8267, 2020-03-16:
@droter, @guiolpei i followed your comment, and i did it. thanks

closed 2020-03-16 by suk8267