realsenseai/realsense-ros #1121: The sample rate of imu and camera

EmiyaEstelle, 2020-03-13:
Hi, I have a Intel realSense T265. I have compiled and installed the ROS package successfully.
I find that the sample rates of IMU and camera are only in a solid values (such as accel 62Hz, gyro 200Hz and fisheye 30Hz). Does the ROS codes support to change the  rate or is the device to support to do this?
I just want to do the calibration with the third tools and sometimes they need the data in a recommending rate. Or are there some official calibration results?

doronhi, 2020-03-17:
The sampling rates are fixed by the camera's capabilities. The device is not capable of producing different rates other then what you can see by typing `rs-enumerate-devices`. The ROS wrapper, realsense2_camera, does not provide such an option, nor does the underlying library librealsense2.
You may need to write your own node to do the interpolation for your needs.

EmiyaEstelle, 2020-03-17:
@doronhi 
Thanks for your anwser. I got that the sample rate is fixed. 
By the way, do you know the official calibration results? I mean that maybe Intel provides it.

doronhi, 2020-03-19:
Sorry, personally I am not aware of any calibration results.

EmiyaEstelle, 2020-03-23:
Thank you all the same！

closed 2020-03-23 by EmiyaEstelle