realsenseai/realsense-ros #1122: Combining D435i with T265

ArvindSanjeev, 2020-03-13:
Hello, I am trying to combine the D435i with T265 to create a robust and accurate SLAM system. I saw it first on this talk from Intel: https://youtu.be/tcJHnHpwCXk?t=1054
Can anyone tell me how they combined the T265 with D435 using Rtabmap? Is this documented somewhere?
P.S: I found this: https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i but it doesn't tell you how to combine it with T265.

dschnabel, 2020-03-13:
I'm no sure if this is how they did in the talk but to get you started you can take a look at my launch files that I linked in this issue which I opened a few days ago (#1120). Before this works you will need to install rtabmap to your system. Since I'm on a Raspberry PI I followed [these steps](https://github.com/introlab/rtabmap/wiki/Installation#raspberrypi). Also note that in my case I'm using the D435, not the D435i.

If you're using a frontend, be sure to enable rviz or rtabmapviz in rovy_rtabmap.launch:
```
<arg name="use_rviz"                  default="true"/>
```

You might be running into the same problem I have with the data from the two cameras not being in sync but if not, then let me know (I'm still waiting if someone has a solution for this).

FPSychotic, 2020-03-15:
> I'm no sure if this is how they did in the talk but to get you started you can take a look at my launch files that I linked in this issue which I opened a few days ago (#1120). Before this works you [...]

Can your RPi4  with both cameras without external powered hub?
Do t265 disconect ramdonly, can you use them lon sesions without t265 crash?
My RPi4 cannot power both without external HUB and t265 disconnect ramdonly by USB physical address conflict.

dschnabel, 2020-03-15:
> Can your RPi4 with both cameras without external powered hub? Do t265 disconect ramdonly, can you use them lon sesions without t265 crash?

Yes, i don't need an external power hub. Both cameras run fine directly from the RPi4 and I haven't encountered any random crash yet. Maybe your power supply isn't powerful enough? I'm using this [voltage regulator](https://www.pololu.com/product/2865) which can deliver up to 6A. For the address conflict, see my [launch file](https://gist.github.com/dschnabel/7d61009f1dd2e1e54400a1443c849ab1#file-rovy_d400_and_t265-launch-L28) where I delay the start of the D435.

But let's not hijack the  discussion here, unless @ArvindSanjeev also wants to run both cameras on a RPi4.

ArvindSanjeev, 2020-03-15:
Thanks for the responses, I am currently agnostic of where I run it, I am using a linux laptop now. @dschnabel will try rtabmap with your launch files. I am still curious about how the two cameras work with each other, do they actually correct each other to build an accurate map jointly? Or are they simply being visualised on rviz independently via rtabmap?

FPSychotic, 2020-03-15:
> Yes, i don't need an external power hub. Both cameras run fine directly from the RPi4 and I haven't encountered any random crash yet. Maybe your power supply isn't powerful enough? I'm using this [...]

I use your same setup with 4A+·3A and I use timed roslanuch too, which avoid the conflict with d435, but not with with any SSD I tried,Coral TPU and just allow connect my lidar in specific sequence and USB port. I ask my self if you have the new board's revision, maybe they fixed silently.
@dschnabel I would not recommend both cameras on RPi4 (and if is possible ARM) as the issues I got many people got too, I could find workarounds but many people couldn't. But even in my case limited my project. really I think RPi4 is not reliable with realsenses (at least my board). For other side on my x86 they work nicely with SSD and TPU included. That proven the limitation is RPi4 (and other ARM SBC too) not the cameras.

dschnabel, 2020-03-15:
@ArvindSanjeev 
> I am still curious about how the two cameras work with each other, do they actually correct each other to build an accurate map jointly? Or are they simply being visualised on rviz independently via rtabmap?

The cameras rather complement each other. The T265 provides the odometry and the D435 provides the image/depth data. Rtabmap combines all data to create a mapping, detect loop closure and estimate location.

@FPSychotic 
> I would not recommend both cameras on RPi4 (and if is possible ARM) as the issues I got many people got too

It's working fine for me (except for the sync problem, but that's not related to power issues I think) because I'm not seeing any disconnects, crashes or restarts.

FPSychotic, 2020-03-15:
> @ArvindSanjeev
>
>
> The cameras rather complement each other. The T265 provides the odometry and the D435 provides the image/depth data. Rtabmap combines all data to create a mapping, detect loop [...]

Mine are  physical USB ID issues that I got workaround. Power issues just don't make it reliable. The only bad of the cameras is t265 you need disconnect it and connect to be recognized. But yeah RPi4 is just crazy, is the same reason do not work kinects for example

ArvindSanjeev, 2020-03-17:
@dschnabel thanks, I was able to get your launch files up and running :)

closed 2020-03-17 by ArvindSanjeev

hzlijianjun, 2021-08-16:
@FPSychotic
With reference to Jetson Nano can't recognize T265 USB issue, I found this code can do power off and on from the post:
_https://forums.developer.nvidia.com/t/usb-power-control/79037/6_

$ wget https://forums.developer.nvidia.com/uploads/short-url/aSZujZz9LQUtN3aS8fuZnByxYtm.zip
$ unzip aSZujZz9LQUtN3aS8fuZnByxYtm.zip
$ sudo apt-get install libusb-1.0-0-dev
$ gcc -o power_cycle realtek_hub_power_cycle.c -lusb-1.0
$ sudo ./power_cycle

This worked perfectly! You can have a try on RPi4.