realsenseai/realsense-ros #1123: Unable to locate package ros-melodic-realsense-camera

S-LR, 2020-03-14:
Hello, everybody, 
I am currently trying to install the ROS wrapper on an NVIDIA Jetson Nano (Ubuntu 18.04 and ROS Melodic) and get the following error message:
E: Unable to locate package ros-melodic-realsense-camera

Does anyone have any idea what could be causing this? 
Many greetings and many thanks
Slr

doronhi, 2020-03-17:
Its "ros-melodic-realsense2-camera", not "ros-melodic-realsense-camera"

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

Hi @S-LR,

As @doronhi have mentioned, the correct name of the package is ros-melodic-realsense2-camera
and you should use that name when you are installing the ROS Wrapper. Please let us know if you need any further clarifications regarding this topic.

Thank you!

S-LR, 2020-03-29:
Thanks a lot and sorry or the late response, everything worked fine now and I think its already corrected in the instructions :)

closed 2020-03-31 by RealSenseCustomerSupport