realsenseai/realsense-ros #1128: Problematic <depend>librealsense2</depend> tag

doisyg, 2020-03-17:
Hi,
The inclusion of the `<depend>librealsense2</depend>` tag in the package.xml is problematic as it triggers the installation of the `ros-melodic-librealsense2` package which is not updated (currently v 2.31) and this forbids the compilation of the realsense2_camera package on an automated system (CI for instance).

doronhi, 2020-03-19:
ros-melodic-librealsense2 is now updated to v2.33.1. It always lags behind the official librealsense2 release.
realsense2_camera depends on librealsense2. Removing this <depend> instruction will cause other issues.
I am open to new suggestions.

doisyg, 2020-03-19:
This is fine as long it is sync with `ros-melodic-realsense2-camera` and that it is usable (which was not the case for the 2.31 version due to usb issues).
Closing

closed 2020-03-19 by doisyg