realsenseai/realsense-ros #1129: "/camera/imu" is not published by D435i.

armanhamzehlou, 2020-03-18:
Hello,

I am following some tutorials on ROS.org:
http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping
http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping
Both of these tutorials point to a topic called "/camera/imu". However, under my published topics, I do not find it:

```
/camera/accel/imu_info
/camera/accel/sample
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/gyro/imu_info
/camera/gyro/sample
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```

Please note that I have updated my Camera's firmware to the latest version since a week ago.
I am running it on an ubuntu client and server environment and trying to pass the data from my Robot server to laptop client so rtabmap on the client can visualize it. However, /camera/imu is not found among topics.

I am using the default rs_camera.launch file along with some parameters using the following command on my server: 

`roslaunch realsense2_camera rs_camera.launch fisheye_fps:=15 depth_fps:=15 infra_fps:=15 color_fps:=15`

I have also seen other users able to get this topic published:
see issue #927 
The above user has been able to get it published using the following command:
`rostopic echo /camera/imu`

If it has been removed from the framework, How can I achieve a remote mapping scenario in an alternative way? (which topic or topics)

ashirsat, 2020-03-18:
Hi,

The imu reading split in two topics and are available in
/camera/accel/sample and /camera/gyro/sample.

Regards,
Aniket Shirsat

doronhi, 2020-03-18:
As @ashirsat said, the device produces 2 streams: accel and gyro.
If you need them united into the same topic you can use the option `unite_imu_method:=linear_interpolation` or `unite_imu_method:=copy`.
Search "unite_imu_method" for further explanation in the README.md file or [here](https://github.com/IntelRealSense/realsense-ros)

armanhamzehlou, 2020-03-18:
> As @ashirsat said, the device produces 2 streams: accel and gyro.
> If you need them united into the same topic you can use the option `unite_imu_method:=linear_interpolation` or [...]

Thank you for your quick response, I followed your advice and read the documentation, I was able to get /camera/imu published using the following command on my server (robot):

`roslaunch realsense2_camera rs_camera.launch fisheye_fps:=15 depth_fps:=15 infra_fps:=15 color_fps:=15  align_depth:=true unite_imu_method:="linear_interpolation"`

I then checked the topics on my client:

[image]
[image]

If I use the compressed version it will throw an exception and asks to use image_transport=compressed. So I ran the rtabmap using the following command on my client:

`roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/aligned_depth_to_color/image_raw rgb_topic:=/camera/color/image_raw/ camera_info_topic:=/camera/color/camera_info approx_sync:=true imu_topic:=/camera/imu queue_size:=100 visual_odometry:=false _image_transport:=compressed
`

and this occures:

[image]

I have already toggle approx_sync to false with no luck.

doronhi, 2020-03-19:
It seems from the `rostopic hz` command's output that frames are not being published on the following topics: `/camera/aligned_depth_to_color/image_raw`, `/camera/color/image_raw`. Can you confirm?
If so, that seems to be the issue. Are they still not being published using rs_camera.launch file without parameters? What is output? What machine and OS do you use?

armanhamzehlou, 2020-03-19:
> It seems from the `rostopic hz` command's output that frames are not being published on the following topics: `/camera/aligned_depth_to_color/image_raw`, `/camera/color/image_raw`. Can you confirm?
> If so, that seems to be the issue. Are they still not being published using rs_camera.launch file without parameters? What is the output? What machine and OS do you use?

You are right. They are not being published. As you see in the image in my previous reply, The topics you mentioned publish on a very irregular base. every 5-6 seconds. 

As for vanilla rs_camera.launch I am attaching the logs of the Server and Client and rostopics command.

**### Server Log:**

<pre><font color="#8AE234"><b>arman@arman-Surface-Pro-6</b></font>:<font color="#729FCF"><b>~</b></font>$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/arman/.ros/log/f7608c3a-69ae-11ea-94a9-2816a86909bd/roslaunch-arman-Surface-Pro-6-1840.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is &lt;1GB.

<b>started roslaunch server http://192.168.0.23:44143/</b>

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 298 lines cut ...]
/camera/color/image_raw     1.257   0.4296      2.186       0.5822    15    
/camera/color/camera_info   29.98   2.861e-06   1.096       0.09127   15    

</pre>

**As you know the camera/imu is not published without further parameters passed to rs_camera.launch.**

**### System Info:**

**For the server I have tried two different environments:**

- A virtual machine running Ubuntu 18.04 with lastest updates as we speak. 8GB of RAM, Connected to USB 3, 2GB of Video memory and 8 cores processor.
-A surface pro 6 running Ubuntu 18.04 with lastest updates as we speak with 16GB of RAM, SSD, Intel Core i7-8650U, Intel UHD Graphics 620.

**For Client:**

- A Virtual Machine with the same specs of the Virtual Server Machine.

I have reversed the situation to see if there is any change by using the surface pro as a client and the VM as the server and still get the same result.

**These VMs are running on a host with:**

- i9 9900k
- RTX 2080Ti 
- 64GB RAM

doronhi, 2020-03-19:
I'd rather leave the rtabmap out of the loop for the moment. 
I don't see any related warnings from the realsense2_camera node. Do you get the same results for "rostopic hz" if you run it and the entire ROS environment (ROS master) on the server alone? I wonder if this is a communication issue..

armanhamzehlou, 2020-03-19:
> I'd rather leave the rtabmap out of the loop for the moment.
> I don't see any related warnings from the realsense2_camera node. Do you get the same results for "rostopic hz" if you run it and the entire ROS environment (ROS master) on the server alone? I wonder if this is a communication issue.

Turns out, You are **absolutely correct**. It does seem like a communication issue. I am doubting my wifi dongle which I have dedicated to the VM.

rostopic hz gave me timely frames on the server not even a single entry missed.

doronhi, 2020-03-20:
I am very glad I could help. Closing the issue then.

closed 2020-03-20 by doronhi