realsenseai/realsense-ros #1130: error: ‘color_sensor’ is not a member of ‘rs2’

jediofgever, 2020-03-18:
instructions that install latest librealsense on ubuntu 18.04 does not work with current develoment branch ;

```
/home/atas/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:704:32: error: ‘color_sensor’ is not a member of ‘rs2’
             else if (sensor.is<rs2::color_sensor>())
                                ^
/home/atas/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:704:32: error: ‘color_sensor’ is not a member of ‘rs2’
/home/atas/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:704:51: error: no matching function for call to ‘rs2::sensor::is()’
             else if (sensor.is<rs2::color_sensor>())
                                                   ^
In file included from /usr/local/include/librealsense2/hpp/rs_device.hpp:8:0,
                 from /usr/local/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/local/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/local/include/librealsense2/rs.hpp:9,
                 from /home/atas/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:11,
                 from /home/atas/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/atas/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/local/include/librealsense2/hpp/rs_sensor.hpp:301:14: note: candidate: template<class T> bool rs2::sensor::is() const
         bool is() const
           
```   ^

MartyG-RealSense, 2020-03-18:
Others experienced this error recently and found that the solution was to compile from source.

https://github.com/IntelRealSense/realsense-ros/issues/1119

jediofgever, 2020-03-18:
Ok I did build from source and fixed but , there should be a warning or some note to avoid users fell into this

MartyG-RealSense, 2020-03-18:
I will tag @doronhi (Intel's developer of the RealSense ROS wrapper) to draw your request to his attention.  Thanks!

closed 2020-03-19 by jediofgever

doronhi, 2020-03-19:
@jediofgever , The issue of building the wrapper with unfitting version of librealsense2 should of course have a proper warning. I'll look into it.
I can also update that librealsense2 v2.33.1 and realsense2_camera version 2.2.13 are now available with ROS distribution for melodic. (`sudo apt-get install ros-melodic-realsense2-camera`. be sure to remove any manual installation you may have to avoid confusion.)

MartyG-RealSense, 2020-03-28:
This case will be closed after the next 7 days  if there are no further responses.  Thanks!