realsenseai/realsense-ros #1132: Unable to launch two cameras at the same time on Nvidia Jetson AGX Xavier

amburkoff, 2020-03-18:
Computer: Jetson Xavier
kernel 4.9.140-tegra
Jetpack 4.2
L4T 32.1.0
ROS Melodic
librealsense 2.32.1
realsense-ros 2.2.13
Firemware: ros camera fw 05.12.02.100

When you try to start two cameras at the same time, Realsense ROS gives an error and starts only one camera. Sometimes, in rare cases, you still manage to launch two.
(see images realsense_multi_camera_1.png and realsense_multi_camera_2.png + log realsense_multi_camera.log)
https://drive.google.com/file/d/1MyXWchXt0wtH1HwFlvkPAfJUxfSBBjP9/view
https://drive.google.com/file/d/1eZU8HvRNljSGNa2EfD_K5nBsqrjo9QgW/view
https://drive.google.com/file/d/1ALH8o3zrJYvLK-8bJ54hw5hJ-Zi5NLpV/view

ashirsat, 2020-03-19:
Could you try and connect both cameras to the usb3 ports only. I had faced
a similar issue when I connected realsense cameras to the usb2 port.

Maybe that will resolve the issue.

Regards,
Aniket Shirsat

doronhi, 2020-03-19:
It could be a problem with the "driver" - RSUSB. I see that it is used, as should be, on the Xavier.
To make the comparison to the NUC clearer, what is the origin of librealsense running on your NUC? Is it built from source or installed from a debian? If the former, what flags did you specify for the `cmake` command?

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

There could be a possibility that the USB3 port  or its power circuitry of your platform not able to provide sufficient power (marginal) to sustain two D435
in the multi-camera configuration white paper, it use 4-port powered USB3 hub (AmazonBasics),
to provide sufficient power for multicamera
https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

hi amburkoff,
Could provided suggestion help on your case?

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

**If we don’t hear from you in 7 days, this issue will be closed**

closed 2020-04-14 by RealSenseCustomerSupport