realsenseai/realsense-ros #1133: Supporting Real Sense ROS node on Windows

roycey, 2020-03-18:
Hello there!

I am from Microsoft and a PM on the Edge Robotics teams. We have ported the RealSense ROS node to work on Windows. We would like it for your team to maintain it going forward and probably release a binary package which would enable your Windows customers to easily consume the package. 

Please reach out to me at rcheeran@microsoft.com and we would love to talk through this. 

Thanks,
Roycey

MartyG-RealSense, 2020-03-18:
Hi @roycey - thanks so much!

I am on the support team of RealSense rather than the developer team, so I cannot speak for the ROS wrapper's developer @doronhi 

However, the stated position of the main RealSense SDK developer team is that they are happy to accept submissions as a pull on the SDK's GitHub, and if it is is successfully integrated then they will try to maintain it from that point onward.  I would imagine that the same principle applies for the RealSense ROS wrapper's GitHub.  @doronhi is the best person to communicate with you about it though.

The RealSense ROS wrapper's GitHub also has a pull page, which can be found at the link below:

https://github.com/IntelRealSense/realsense-ros/pulls

Thanks again for supporting RealSense ROS!

MartyG-RealSense, 2020-04-09:
I will tag @doronhi again to re-highlight your request to him.

doronhi, 2020-04-17:
Thanks for #1147. We will address it as soon as possible, considering our priorities and schedule.

MartyG-RealSense, 2020-04-20:
Closed due to response to @roycey  from the RealSense ROS developer having been made.  The pull  https://github.com/IntelRealSense/realsense-ros/pull/1147 remains open.

closed 2020-04-20 by MartyG-RealSense