realsenseai/realsense-ros #1136: unable to locate package ros-melodic-realsense-camera

PonyZhou, 2020-03-21:
run  "sudo apt-get  install  ros-melodic-realsense-camera"  command and got a error "package ros-melodic-realsense-camera"
i have updated the source list but still got the error

MartyG-RealSense, 2020-03-21:
I note that in the instruction you quoted, there is a '2' missing from the 'realsense' part of the instruction.  It should be:

sudo apt-get install ros-melodic-realsense2-camera

PonyZhou, 2020-03-21:
Thanks！You are right and had better correct the same wrong command in the Installation Instructions.

MartyG-RealSense, 2020-03-21:
@PonyZhou  If you are referring to this instruction:

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

I will tag the wrapper's developer @doronhi into this discussion to highlight about including the installation instruction for Melodic.  Thanks!

PonyZhou, 2020-03-21:
Yes, under the "Method 1: The ROS distribution:"

doronhi, 2020-03-22:
Fixed. Thank you guys!

MartyG-RealSense, 2020-04-09:
I will close this case now that the fix has been performed.  Thanks!

closed 2020-04-09 by MartyG-RealSense