realsenseai/realsense-ros #1137: How to get corrected IMU data?

stevemartinov, 2020-03-21:
Hi,

I am using D435i and I am trying to get the IMU in a correct way. As I understood . IMU is based in depth frame which has different coordinate system and I spend the last 2 days to get the right orientation so whenever I put my camera in a normal position (not facing downwards) I get the correct yaw. I tried to use Roll from the camera as Yaw (as if the camera is facing down the roll position is becoming Yaw position when the camera is facing forward) but no luck.

Can you guys tell me what do I need to do to use IMU from the camera?

Here is my launch file (note that the whole TF of realsense is attached to `camera_link` which has no rotation):

 ```
<arg name="offline"          default="false"/>
    <include unless="$(arg offline)" 
        file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="align_depth" value="true"/>
        <arg name="linear_accel_cov" value="1.0"/>
        <arg name="unite_imu_method" value="linear_interpolation"/>
        <arg name="enable_depth" value="false"/>
        <arg name="enable_infra2" value="false"/>
        <arg name="publish_odom_tf" value="false"/>
        <arg name="gyro_fps" value="200"/>
        <arg name="accel_fps" value="63"/>
    </include>
    
    <node pkg="imu_filter_madgwick" type="imu_filter_node" name="ImuFilter">
        <param name="use_mag" type="bool" value="false" />
        <param name="publish_tf" type="bool" value="true" />
        <param name="world_frame" type="string" value="enu" />
        <remap from="/imu/data_raw" to="/camera/imu"/>
    </node>
```

MartyG-RealSense, 2020-03-21:
Information about the axis system of the D435i's IMU can be found at the link below:

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system

As the D435i IMU does not have internal calibration, it is recommended that a provided IMU calibration software tool is used on the IMU.  This is separate from the Dynamic Calibrator software used to calibrate the non-IMU sensors of the camera.

https://github.com/IntelRealSense/librealsense/tree/development/tools/rs-imu-calibration#rs-imu-calibration-tool

Intel has a guidance document on performing this calibration.  It relies on a six-stage calibration process of starting the calibration from the facing-forwards position and then rotating it on its sides and pointing it upwards and downwards.

https://dev.intelrealsense.com/docs/depth-camera-d435i-imu-calibration

If you are aiming to calibrate the D435i for the downward facing position, the advice in the link below will hopefully be helpful.

https://github.com/IntelRealSense/librealsense/issues/4751#issuecomment-526102539

In regard to using the D435i and its IMU with ROS, Intel has published a SLAM setup guide for D435i:

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

stevemartinov, 2020-03-21:
Hi, this is not helpful at all. I was not asking about the axis (as I stated that the axis are known) or about the calibration. I am asking how can I convert the depth axis (IMU axis) to a regular ROS axis? From the code I found out that you are just rotating pitch and yaw by -PI/2 and when I am getting the data, I am just adding PI/2 to both pitch and yaw respectively and making yaw to be a pitch and roll to be my yaw but I think I might be doing wrong as I do not always get the desired results

MartyG-RealSense, 2020-03-21:
The ROS wrapper documentation states the following information:

**odom_frame_id** defines the origin coordinate system in ROS convention (X-Forward, Y-Left, Z-Up).  pose topic defines the pose relative to that system.

MartyG-RealSense, 2020-04-09:
This case will be closed after 7 days from the time of writing this message if there are no further comments.   Thanks!

closed 2020-04-15 by MartyG-RealSense

shuhaowu, 2020-10-25:
Isn't `odom_frame_id` for the T265 built-in tracking? This is about the coordinate system of the imu data.

MartyG-RealSense, 2020-10-25:
@shuhaowu   Thanks for your question. **odom_frame_id** is not a concept that is exclusive to the T265's IMU.