realsenseai/realsense-ros #1138: Noisy pointcloud

stevemartinov, 2020-03-22:
Hi,

Is this a normal behaviour when I enable pointcloud on my D435i? Attaching a video:

https://drive.google.com/file/d/1n1ehSkeoSgNz64-yUzq4lAeNCsIVQvWg/view?usp=sharing

MartyG-RealSense, 2020-03-24:
It is difficult to provide a diagnosis from the video.  the depth information does appear to be fluctuating and distorting to a highly unusual extent that does not look correct.  

It also resembles a previously reported 'elongated point cloud' effect where the point cloud appears to distort when objects are viewed from side-on.  Links to these cases are provided below.  Hopefully you will find helpful advice in them.

https://github.com/IntelRealSense/librealsense/issues/3741

https://github.com/IntelRealSense/librealsense/issues/4553

https://forums.intel.com/s/question/0D50P00004IwgfKSAR/the-depth-quality-get-worse-in-the-edge-of-object

Are you using any post-processing on the point cloud please?  Page 3, figure 2 of Intel's post-processing white paper document says that although a post-processing filter that provides smoothing can smooth out edges, it can also cause the edges to become elongated, rounded or overshoot.

https://www.intel.com/content/www/us/en/support/articles/000028866/emerging-technologies/intel-realsense-technology.html

ilkaydost, 2020-03-26:
Hi MartyG-RealSense,

I have same problem too. While I capturing data it was to noise and I couldn't handle it. I tried several methods like as changing decimation filter, adjusting depth units, exposure , golden factory reset etc. All these methods didn't worked. I'm also searching for new answers.

1-)How can I convert depth(raw) image to point cloud(pcd). Because depth image seems to me good.

2-)Are there any json file to make best visual settings in 4-5 meters.

------------
Depth image;
[image]
Point Cloud - (I converted with ply2pcd) ;
[image]
[image]

-----------
These two, I captured with ros record

[image]
[image]

MartyG-RealSense, 2020-03-27:
@ilkaydost  Yes, I agree that your depth image is good overall.

In the sections on the depth image where there are black areas on the floor, please check whether there are reflections or strong spots of light illumination in those areas that may make it hard for the camera to read the depth detail in those areas.

In regard to your questions:

1.  To get a .pcd format file, I believe that the process is to convert the RealSense point cloud to PCL format, and then use PCL to convert to a PCD file.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl

The PCL wrapper has sample programs called **rs-pcl** and **rs-pcl-color** for doing the RealSense to PCL data conversion for a point cloud.

 2.  The documentation for Visual Presets (another name for a JSON) recommends the JSON file called "Default" for a good balance of values, including best visual appeal, clean edges and reduced point cloud spraying.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table

ilkaydost, 2020-03-28:
@MartyG-RealSense Thank you for responce, 

Actually, I have problem with also flat surface. In the picture above flat surface not steady. This cause problem when I made segmentation. I tried rs-pcl and rs-pcl-color that you mentioned your reponse but result hasn't changed. I'm wondering, are there any convertion .raw to pcd ?  or how can I improve my point cloud data? Because when I capturing with Kinect I didn't faced these problems. 
  
In edit section you mentioned, I captured ply and I convert with cloud compare but same problems going through.

These are pictures in rviz, realsense- viewer and rs-pcl;

Rviz data:
[image]
-----
Realsense- viewer data with post processing:

In this picture there are white wall but when I look more closely you can see noise even small distance !

[image]
[image]
----

Rs-pcl:

[image]

MartyG-RealSense, 2020-03-28:
Kinect cameras use a different camera sensing technology to the RealSense 400 Series.  Kinect V1 uses Structured Light (like the RealSense SR300 / SR305) and Kinect V2 uses Time of Flight (which has similarities to the forthcoming new RealSense L515 lidar-based depth camera).  The 400 Series cameras are Stereo based (constructing a depth image from the left and right IR imagers, hence 'stereo').

I recall a past case discussing "wavy" surfaces.  It was suggested that the waviness may increase as the camera gets further away from the object, so moving the camera closer may help if that is possible.  The Chief Technical Officer of the RealSense group at Intel (agrunnet) suggests in this discussion that if the field of view at close range becomes a problem then more than one camera could be combined to widen the overall field of view.

https://github.com/IntelRealSense/librealsense/issues/1375

My recollection is that conversions of .raw data tend to be awkward.  A RealSense user did though post a script for converting .raw data into a .ply, which is at least a step towards getting a PCD.

https://github.com/IntelRealSense/librealsense/issues/3286#issuecomment-465648074

stevemartinov, 2020-03-28:
I am sorry to say that but are you guys seriously suggesting to simply buy one more camera to cope with a problem? This is utterly unacceptable. 
People are buying your end product for $200 without knowing that they will face such problems and spending weeks to simply fix some non-solvable driver problems and the best answer they can get is to either "unplug and plug the sensor" or "buy another camera suggested by our CTO". I am literally done with Realsense products as I tried Orbbec cameras and they do not have such problems at all!

closed 2020-03-28 by stevemartinov

RalphCodesTheInternet, 2021-08-11:
Agree... Facing the same problem with L515, and the range is not even close to what was advertised..

MartyG-RealSense, 2021-08-11:
Hi @RalphCodesTheInternet  The L515 supports a range of up to 9 meters.  However, optimal depth sensing results may be achieved at 4 meters and under.  This is similar to the 400 Series cameras, where accuracy can begin to drift noticably beyond 3 meters (though the D455 model has the same accuracy at 6 meters that the D435 / D435i do at 3 meters).

sunwell1994, 2022-03-30:
> Agree... Facing the same problem with L515, and the range is not even close to what was advertised..

For L515, the detectable distance is related to reflectivity. The max range requires a large reflectivity.