realsenseai/realsense-ros #1139: Latest realsense-ros release error

caiobarrosv, 2020-03-22:
Hello all,

**ros-version**: Kinetic | Ubuntu 16.04

Even though the latest realsense-ros release says that it is compatible with LibRealSense v2.33.1, the 
[CMakeList.txt](https://github.com/IntelRealSense/realsense-ros/blob/2.2.13/realsense2_camera/CMakeLists.txt#L37-L40) requires the 2.32.1 version.

I Installed the LibRealSense v2.33.1 as stated [here](https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.13) but I am getting the following error:

```bash
CMakeLists.txt:37 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.32.1".

  The following configuration files were considered but not accepted:

    /opt/ros/kinetic/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Config.cmake, version: 2.31.0

CMake Error at /home/vitu/real-time-grasp/src/realsense-ros/realsense2_camera/CMakeLists.txt:39 (message):

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
```

The command sudo apt list | grep libreal returns the realsense packages already installed:

```
librealsense2/xenial,now 2.33.1-0~realsense0.2078 amd64 [installed]
librealsense2-dkms/xenial,now 1.3.11-0ubuntu1 all [installed]
librealsense2-gl/xenial,now 2.33.1-0~realsense0.2078 amd64 [installed]
librealsense2-udev-rules/xenial,now 2.33.1-0~realsense0.2078 amd64 [installed]
librealsense2-utils/xenial,now 2.33.1-0~realsense0.2078 amd64 [installed]
ros-kinetic-librealsense2/xenial,now 2.31.0-1xenial-20191214-001452+0000 amd64 [installed]
```

What am I missing here?

Thank you in advance.

ps: I don't have the camera available at the moment.

doronhi, 2020-04-02:
you have "ros-kinetic-librealsense2/xenial,now 2.31.0..." installed. Remove it and try again.

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

Hi @caiobarrosv,

Have you tried to remove all the versions of ROS Wrapper you already have and install the latest version from scrtach?

Thank you!

caiobarrosv, 2020-04-13:
Thank you for the suggestion @doronhi & @RealSenseCustomerSupport. I installed the realsense-ros 2.2.11 and librealsense  2.31.0 from source. These versions seem to work well with my project.

closed 2020-04-13 by caiobarrosv