realsenseai/realsense-ros #1140: Infra1 and Infra2 topics - what kind of images ?

ninamwa, 2020-03-23:
Hi.
A little (maybe) off-topic question, but I am wondering what images are published to the topics:
Rectified infra1 image: /camera/infra1/image_rect_raw
Rectified infra2 image: /camera/infra2/image_rect_raw

Is this the images from the Left and Right Imager on the camera (HD image sensors) or is it from the IR projector?
I guess I am confused because they are named infra, so it sounds like it is an image from the infrared projector, but at the same time, there are two of them?

I tried to look for documentation of this somewhere, but I could not find any. Thanks

MartyG-RealSense, 2020-03-24:
Infra1 = left infraed imager.  Infra2 = right infrared imager.

The IR projector is a separate component from the left and right IR imagers that is responsible for projecting a semi-random dot pattern onto a scene.  This pattern can be used to help the camera analyse flat / low-texture surfaces such as walls, doors and desks for depth detail.  If the scene is well-lit though then the camera can alternatively use ambient light in the scene to analyse depth instead of using a projected dot pattern.

ninamwa, 2020-03-24:
Thank you for your answer :-) Just to make sure I understood you right: The left infrared imager you mention is the same as the left imager in the picture (p.34 in the datasheet of D400 Series Product Family)?
I can not find anywhere in the datasheet that these imagers were based on IR, so that is what confuses me. 

[image]

MartyG-RealSense, 2020-03-24:
The 400 Series cameras construct their stereo depth image from the left and right IR imagers.    This is why there is not a separate 'depth imager' component on the camera.

Yes, the left and right imagers are IR imagers.  On page 35 and 36, just past that diagram, their specification is listed under the heading 'Wide Left And Right Imager Properties'.

It is called 'wide' because the IR imager component on the D420 and D430 cameras is wider than the different OmniVision imager component used on the D400, 410 and 415 cameras.

ninamwa, 2020-03-24:
Thank you very much for your explanation!

closed 2020-03-24 by ninamwa