realsenseai/realsense-ros #1142: Upgraded librealsense, now getting errors

poorboy44, 2020-03-23:
I upgraded librealsense to `2.33.1-0~realsense0.2079`. I am running  Ubuntu 18.04 with Linux kernel version `5.3.0-42-generic`. 

Now when I run the ROS integration, I get errors related to `backend-v4l2`. Can you advise?

```
(base) poorboy44@poorboy44-XPS-8930:~/catkin_ws$ roslaunch realsense2_camera rs_multiple_devices_sn.launch serial_no_camera1:=943222072680 serial_no_camera2:=944122073572
... logging to /home/poorboy44/.ros/log/dd9ebcf2-6d5c-11ea-b811-5cea1d90be79/roslaunch-poorboy44-XPS-8930-4786.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://poorboy44-XPS-8930:44911/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 15
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/depth_fps: 15
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: True
 * /camera1/realsense2_camera/enable_fisheye2: True
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_pointcloud: True
 * /camera1/realsense2_camera/enable_sync: True
 * /camera1/realsense2_camera/filters: 
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: 0
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: 0
 * /camera1/realsense2_camera/gyro_fps: 0
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
 * /camera1/realsense2_camera/imu_optical_frame_id: camera1_imu_optic...
 * /camera1/realsense2_camera/infra1_frame_id: camera1_infra1_frame
 * /camera1/realsense2_camera/infra1_optical_frame_id: camera1_infra1_op...
 * /camera1/realsense2_camera/infra2_frame_id: camera1_infra2_frame
[... 257 lines cut ...]
 23/03 17:20:23,048 WARNING [140691489986304] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
 23/03 17:20:23,048 ERROR [140691489986304] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 18
 23/03 17:20:23,048 WARNING [140691489986304] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
 23/03 17:20:23,048 ERROR [140691489986304] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 18
 23/03 17:20:23,048 WARNING [140691489986304] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
 23/03 17:20:23,048 ERROR [140691489986304] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 18
 23/03 17:20:23,048 WARNING [140691489986304] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
 23/03 17:20:23,052 ERROR [140442565637888] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 18
 23/03 17:20:23,052 WARNING [140442565637888] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
 23/03 17:20:23,052 ERROR [140442565637888] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 18
 23/03 17:20:23,052 WARNING [140442565637888] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
 23/03 17:20:23,052 ERROR [140442565637888] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 18
 23/03 17:20:23,052 WARNING [140442565637888] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
 23/03 17:20:23,052 ERROR [140442565637888] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 18
 23/03 17:20:23,052 WARNING [140442565637888] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
^C[camera2/realsense2_camera-5] killing on exit
[camera2/realsense2_camera_manager-4] killing on exit
[camera1/realsense2_camera-3] killing on exit
[camera1/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

poorboy44, 2020-03-24:
I'd also add that now the process seems to completely overwhelm my CPU and make my computer unusable. Something seems quite wrong ...

doronhi, 2020-03-25:
Can you see the same behavior when running a single camera? Can you see that using realsense-viewer?

poorboy44, 2020-03-25:
I get the same behavior with a single camera. 

It's hard to check with `realsense-viewer`, because launching it freezes up my computer.

poorboy44, 2020-03-25:
I’m running the most recent firmware, librealsense2 (Debian package) and realsense-ROS library (built from source).

doronhi, 2020-03-26:
It's a kernel/librealsense2 issue that needs investigating.
Is that problem reproducible?
Could you tell if it happens on other kernels as well?

poorboy44, 2020-03-26:
Unfortunately I only have the one computer, so it's hard for me to check other kernels...

doronhi, 2020-04-02:
@ev-mp , this seems like a librealsense2/kernel issue. Any thoughts?

mikeNike9, 2020-04-06:
I am also dealinbg with this same thing. I have done all the same things as stated above. 
I actually get a new error on the frame callback.

mikeNike9, 2020-04-06:
Hello

I figured something our on our end. 
We were using the same kernal version as you but the only thing different was my Intel d415 firmware was using Pre released firmware.
I simply put my firm back to a stable version and did not see the backend-v4l2.h error.
hope this helps

poorboy44, 2020-04-06:
What is the firmware version?

mikeNike9, 2020-04-06:
i am using 5.12.01

But we still are getting the same error.
Have you been able to fix this issue on your side. 
@poorboy44

poorboy44, 2020-04-06:
I haven’t...

mikeNike9, 2020-04-07:
@doronhi 

Is there anything we can do to get this to work.
Maybe use a different version of librealsense?
I am trying to deploy an application and this is getting in the way of me doing so.

doronhi, 2020-04-07:
I still don't have a lead yet. Can anyone check if it appears with a librealsense application as well, such as realsense-viewer or rs-data-collect? I'm trying to establish if this is a ROS wrapper issue or a librealsense one.

poorboy44, 2020-04-07:
Are you not able to reproduce the issue with the latest librealsense/firmware/ROS wrapper?

haydenhager, 2020-04-07:
@doronhi I am working with @mikeNike9 on this project. We tested with librealsense-viewer and had the same issue occur.

mikeNike9, 2020-04-07:
@doronhi 
Do you have and ETA of the patch. 
If this is going to take a while then I think we will need to move in different direction.

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

Hi all,

In order to assist you in the most efficient manner, we need more information about the issue.
Please provide us with the following details about your system configuration the issue could be reproduced  on:
ROS distribution version
HW configuration details (platforms names, how the camera(s) is/are connected, etc)

Along with that information, please share with us Intel.RealSense.Viewer logs and describe the exact scenario that leads to the problem.

Thank you!

haydenhager, 2020-04-09:
Intel Celeron J1900
Using ROS Melodic 2.2.13
realsense-ros installed using Method 2
Installed librealsense using apt-get
Ubuntu 18.04.1
Kernel 5.3.0-45-generic x86-64
D415 Firmware 05.12.03.00
SDK 2.33.1
I have one camera installed via supplied usb3 cable to usb 3 on machine.

Output from realsense-viewer log:
 09/04 09:45:20,733 WARNING [140066563274496] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
 09/04 09:45:20,738 ERROR [140066563274496] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 13
 09/04 09:45:20,738 WARNING [140066563274496] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 13
 09/04 09:45:20,738 WARNING [140066563274496] (backend-v4l2.h:203) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 13
 09/04 09:45:20,738 ERROR [140066563274496] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 13
 09/04 09:45:20,738 WARNING [140066563274496] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 13
 09/04 09:45:20,738 WARNING [140066563274496] (backend-v4l2.h:203) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 13
 09/04 09:45:20,738 ERROR [140066563274496] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 13
 09/04 09:45:20,738 WARNING [140066563274496] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 13
 09/04 09:45:20,738 WARNING [140066563274496] (backend-v4l2.h:203) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 13
 09/04 09:45:20,738 ERROR [140066563274496] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 13
 09/04 09:45:20,738 WARNING [140066563274496] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 13
 09/04 09:45:21,865 WARNING [140066332604160] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
 09/04 09:45:21,885 ERROR [140066332604160] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 21
 09/04 09:45:21,885 WARNING [140066332604160] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 21
 09/04 09:45:21,885 WARNING [140066332604160] (backend-v4l2.h:203) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 21
 09/04 09:45:21,885 ERROR [140066332604160] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 21
 09/04 09:45:21,885 WARNING [140066332604160] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 21
 09/04 09:45:21,885 WARNING [140066332604160] (backend-v4l2.h:203) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 21
 09/04 09:45:21,885 ERROR [140066332604160] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 21
 09/04 09:45:21,885 WARNING [140066332604160] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 21
 09/04 09:45:21,885 WARNING [140066332604160] (backend-v4l2.h:203) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 21
 09/04 09:45:21,885 ERROR [140066332604160] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 21
 09/04 09:45:21,886 WARNING [140066332604160] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 21
 09/04 09:45:21,904 ERROR [140066332604160] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 21
 09/04 09:45:21,904 WARNING [140066332604160] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 21
 09/04 09:45:21,904 ERROR [140066332604160] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 21
 09/04 09:45:21,905 WARNING [140066332604160] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 21
 09/04 09:45:21,905 ERROR [140066332604160] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 21
 09/04 09:45:21,905 WARNING [140066332604160] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 21
 09/04 09:45:21,905 ERROR [140066332604160] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 21
 09/04 09:45:21,905 WARNING [140066332604160] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 21

Output from rs_aligned_depth:
[ INFO] [1586442992.628143420]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[ INFO] [1586442992.628189633]: Device with name Intel RealSense D415 was found.
[ INFO] [1586442992.629382295]: Device with port number 2-1 was found.
[ INFO] [1586442992.637120984]: getParameters...
[ INFO] [1586442992.802402988]: setupDevice...
[ INFO] [1586442992.802489713]: JSON file is not provided
[ INFO] [1586442992.802536538]: ROS Node Namespace: camera
[ INFO] [1586442992.802590131]: Device Name: Intel RealSense D415
[ INFO] [1586442992.802639475]: Device Serial No: 821212062344
[ INFO] [1586442992.802723801]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[ INFO] [1586442992.802774874]: Device FW version: 05.12.03.00
[ INFO] [1586442992.802823870]: Device Product ID: 0x0AD3
[ INFO] [1586442992.802871679]: Enable PointCloud: Off
[ INFO] [1586442992.802919068]: Align Depth: On
[ INFO] [1586442992.802980893]: Sync Mode: On
[ INFO] [1586442992.803189540]: Device Sensors: 
[ INFO] [1586442992.803305486]: Stereo Module was found.
[ INFO] [1586442992.803378075]: RGB Camera was found.
[ INFO] [1586442992.803489064]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1586442992.803537941]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[... cut ...]
 09/04 09:36:33,770 WARNING [139973293553408] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 19

These logs are from the first time I got on this morning to run both of these programs. 

In addition, I upgraded librealsense to 2.34.0 and am getting the same error with less frequency in realsense viewer, but still occurring.

Also, another error: "ERROR [140456452417280]  (types.h:307) get_device_time_ms() took too long (more than 2 mSecs)
This error is occurring frequently but doesn't appear to have an effect on usability. .

poorboy44, 2020-04-09:
I upgraded to the latest librealsense debian packages, and seem to be unable to reproduce the issue so far. Here's what I have installed:
```
poorboy44@poorboy44-XPS-8930:~/catkin_ws$ sudo apt list --installed | grep librealsense

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

librealsense2/bionic,now 2.34.0-0~realsense0.2207 amd64 [installed]
librealsense2-dbg/bionic,now 2.34.0-0~realsense0.2207 amd64 [installed]
librealsense2-dev/bionic,now 2.34.0-0~realsense0.2207 amd64 [installed]
librealsense2-dkms/bionic,now 1.3.13-0ubuntu1 all [installed]
librealsense2-gl/bionic,now 2.34.0-0~realsense0.2207 amd64 [installed,automatic]
librealsense2-net/bionic,now 2.34.0-0~realsense0.2207 amd64 [installed,automatic]
librealsense2-udev-rules/bionic,now 2.33.1-0~realsense0.2079 amd64 [installed,upgradable to: 2.34.0-0~realsense0.2207]
librealsense2-utils/bionic,now 2.34.0-0~realsense0.2207 amd64 [installed]
ros-melodic-librealsense2/bionic,now 2.33.1-2bionic.20200309.152858 amd64 [installed]
```

poorboy44, 2020-04-09:
Interestingly, the ROS logs indicate that it's still using `LibRealSense v2.33.1`. I notice I have a package called `ros-melodic-librealsense2` which is version 2.33.1. Also, I seem to have a package `librealsense2-dkms` but this version (1.3.13) is not consistent with the other librealsense packages (2.34.0). I guess I don't understand the relationship between these packages and ROS.

```
poorboy44@poorboy44-XPS-8930:~/catkin_ws$ roslaunch realsense2_camera rs_multiple_devices_sn.launch serial_no_camera1:=943222072680 serial_no_camera2:=944122073572
... logging to /home/poorboy44/.ros/log/c6a50380-7a81-11ea-9399-5cea1d90be79/roslaunch-poorboy44-XPS-8930-21930.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://poorboy44-XPS-8930:41129/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: True
 * /camera1/realsense2_camera/enable_fisheye2: True
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_pointcloud: True
 * /camera1/realsense2_camera/enable_sync: True
 * /camera1/realsense2_camera/filters: 
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: 0
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: 0
 * /camera1/realsense2_camera/gyro_fps: 0
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
 * /camera1/realsense2_camera/imu_optical_frame_id: camera1_imu_optic...
 * /camera1/realsense2_camera/infra1_frame_id: camera1_infra1_frame
 * /camera1/realsense2_camera/infra1_optical_frame_id: camera1_infra1_op...
 * /camera1/realsense2_camera/infra2_frame_id: camera1_infra2_frame
 * /camera1/realsense2_camera/infra2_optical_frame_id: camera1_infra2_op...
 * /camera1/realsense2_camera/infra_fps: 30
[... 186 lines cut ...]
[ INFO] [1586450840.382953662]: Done Setting Dynamic reconfig parameters.
[ INFO] [1586450840.385637251]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1586450840.389964498]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1586450840.390015456]: setupPublishers...
[ INFO] [1586450840.392666716]: Expected frequency for depth = 30.00000
[ INFO] [1586450840.395320825]: Expected frequency for color = 30.00000
[ INFO] [1586450840.412492402]: setupStreams...
[ INFO] [1586450840.420080884]: Expected frequency for color = 30.00000
[ INFO] [1586450840.429990881]: setupStreams...
[ INFO] [1586450840.430593317]: insert Depth to Stereo Module
[ INFO] [1586450840.430614509]: insert Color to RGB Camera
[ INFO] [1586450840.450302068]: SELECTED BASE:Depth, 0
[ INFO] [1586450840.454856978]: RealSense Node Is Up!
[ INFO] [1586450840.463474607]: insert Depth to Stereo Module
[ INFO] [1586450840.463585745]: insert Color to RGB Camera
[ INFO] [1586450840.488362448]: SELECTED BASE:Depth, 0
[ INFO] [1586450840.513095689]: RealSense Node Is Up!
 09/04 10:47:20,791 ERROR [140418731575040] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
[... cut ...]
 09/04 10:47:20,943 ERROR [140418731575040] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 09/04 10:47:20,981 ERROR [140023852693248] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 09/04 10:47:21,183 ERROR [140023852693248] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 09/04 10:47:21,196 ERROR [140418731575040] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 09/04 10:47:21,485 ERROR [140023852693248] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
```

haydenhager, 2020-04-10:
@poorboy44  I was able to update all the debian packages and that seems to have fixed the v412.h backend error. We now have new errors that we will post in a new issue. 

Thanks!

RealSenseCustomerSupport, 2020-04-14:
----------------------------------------------

@haydenhager, Thank you for the provided details.
@poorboy44, If the original problem with backend-v4l2 was solved by upgrading to the latest LRS Debian packages and you don't need any further assistance with this matter, let's close this issue and discuss other problems in separate threads.

Thank you!

closed 2020-04-20 by RealSenseCustomerSupport