realsenseai/realsense-ros #1143: i don't know about subscribe pcl topic

suk8267, 2020-03-24:
i followed tutorial of realsense camera, so i published to rviz. but when i got topic by subscriber, i have to include msg file but in msg folder have Extrinsics.msg and IMUinfo.msg .   
i think IMUinfo is for accelerator and gyro sensor. Extrinsic is may be for Rotation of camera. so where is msg file for pointcloud topic?

MartyG-RealSense, 2020-03-25:
Could you provide a link to the tutorial that you were following, please?   Thanks!

suk8267, 2020-03-26:
i followed "https://github.com/IntelRealSense/realsense-ros" here in the code tutorial.

MartyG-RealSense, 2020-03-26:
The documentation says:. 

*********

The D435i and T265 cameras have built in IMU components which produce 2 unrelated streams: gyro - which shows angular velocity and accel which shows linear acceleration.  Each with it's own frequency.  By default, 2 corresponding topics are available, each with only the relevant fields of the message sensor_msgs::Imu are filled out. 

Setting unite_imu_method [in the launch instruction] creates a new topic, imu, that replaces the default gyro and accel topics.  Under the new topic, all the fields in the Imu message are filled out.

****************
So as far as I can tell (I do not currently have RViz set up to test this), there are only meant to be two msg categories.

suk8267, 2020-03-27:
@MartyG-RealSense thanks for your help. so i tried this code just get the step of point. but it took this error do you know why this happen?

#include "ros/ros.h"
#include "sensor_msgs/PointCloud2.h"

void chatterCallback(const realsense2_camera::PointCloud2::ConstPtr& pcl){
        ROS_INFO("point_step : %d",pcl->point_step);
}

int main(int argc, char **argv){
        ros::init(argc, argv, "icp_sub");

	ros::NodeHadle nh;

	ros::Subscriber sub = nh.subscribe("/camera/depth_registered/points", 10000, chatterCallback);

        ros::spin();

        return 0;
}
//////////////////error///////////////
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/icp.cpp:4:28: error: ‘realsense2_camera’ does not name a type
 void chatterCallback(const realsense2_camera::PointCloud2::ConstPtr& pcl){
                            ^
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/icp.cpp:4:58: error: expected unqualified-id before ‘::’ token
 void chatterCallback(const realsense2_camera::PointCloud2::ConstPtr& pcl){
                                                          ^
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/icp.cpp:4:58: error: expected ‘)’ before ‘::’ token
/home/hongseok/catkin_ws/src/realsense-ros/realsense2_camera/src/icp.cpp:4:58: error: expected initializer before ‘::’ token
realsense-ros/realsense2_camera/CMakeFiles/icp_sub.dir/build.make:62: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/icp_sub.dir/src/icp.cpp.o' failed
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/icp_sub.dir/src/icp.cpp.o] Error 1
CMakeFiles/Makefile2:3265: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/icp_sub.dir/all' failed
make[1]: *** [realsense-ros/realsense2_camera/CMakeFiles/icp_sub.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

thanks

MartyG-RealSense, 2020-03-27:
My research indicates that the "does not name a type" error may occur because something (in this case 'realsense2_camera') has not been defined and so the program does not know it exists.

I would recommend checking the includes as a first step to see if something vital is not being imported.

closed 2020-04-15 by MartyG-RealSense