realsenseai/realsense-ros #1144: Possible memory leak

abylikhsanov, 2020-03-24:
Hi,

I have upgraded both librealsense and ros wrapper to the latest release and I think some memory leak has been introduced. Whenever I run nodelet manager, ater N minutes it dies. It is quite hard to catch that moment and I tried to run with Valgrind. Unfortunately, I was not able to catch when the process dies but here is the latest log:

```
==789== 
==789== Process terminating with default action of signal 15 (SIGTERM)
==789==    at 0x77219F3: futex_wait_cancelable (futex-internal.h:88)
==789==    by 0x77219F3: __pthread_cond_wait_common (pthread_cond_wait.c:502)
==789==    by 0x77219F3: pthread_cond_wait@@GLIBC_2.3.2 (pthread_cond_wait.c:655)
==789==    by 0x560AFCE: ros::ROSOutAppender::logThread() (in /opt/ros/melodic/lib/libroscpp.so)
==789==    by 0x7500BCC: ??? (in /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1)
==789==    by 0x771B6DA: start_thread (pthread_create.c:463)
==789==    by 0x69B788E: clone (clone.S:95)
==789== 
==789== HEAP SUMMARY:
==789==     in use at exit: 8,478,798 bytes in 609 blocks
==789==   total heap usage: 34,020 allocs, 33,411 frees, 10,422,235 bytes allocated
==789== 
==789== LEAK SUMMARY:
==789==    definitely lost: 96 bytes in 1 blocks
==789==    indirectly lost: 0 bytes in 0 blocks
==789==      possibly lost: 1,520 bytes in 5 blocks
==789==    still reachable: 8,477,182 bytes in 603 blocks
==789==         suppressed: 0 bytes in 0 blocks
==789== Rerun with --leak-check=full to see details of leaked memory
==789== 
==789== For counts of detected and suppressed errors, rerun with: -v
==789== Use --track-origins=yes to see where uninitialised values come from
==789== ERROR SUMMARY: 2 errors from 1 contexts (suppressed: 0 from 0)
```

doronhi, 2020-03-25:
Can you please specify what system you are running on? Did you install both debian packages or built any one of them from source?
Also, could you specify for the record what versions of librealsense2 and realsense2_camera you are using?

RealSenseSupport, 2020-08-14:
Hi,

Will you be needing further help with this? Are you still experiencing the same thing on latest?

If we don’t hear from you in 7 days, this issue will be closed.

Thanks