realsenseai/realsense-ros #1145: Nvidia Jetson TX2 with Realsense D435 Crashing

cdimmig, 2020-03-24:
| Component | Version |
|----------------------|--------------------------------------------|
| Platform | Nvidia Jetson TX2 Jetpack 3.3 CTI-L4T-V121 |
| Operating System | Ubuntu 16.04 |
| Kernel | Linux 4.4.38-tegra |
| Camera Model | D435 |
| librealsense | 2.28.1 |
| realsense_ros | 2.2.8 |
| ddynamic_reconfigure | 0.2.0 |

Hello,

I am working with a freshly flashed TX2 with the above configuration for working with a Realsense D435.  The D435 has been updated to the "recommended firmware" via the realsense_viewer GUI. I am seeing the camera crash intermittently (every few runs) with one of the following errors:
- [ WARN] [1585059250.961663106]: Hardware Notification:USB CAM overflow,1.58506e+12,Error,Hardware Error
- [ WARN] [1585063678.744642948]: Hardware Notification:IR stream start failure,1.58506e+12,Error,Hardware Error
[ERROR] [1585063678.745066533]: Hardware Reset is needed. use option: initial_reset:=true
[ERROR] [1585063681.246201738]: The device has been disconnected!
[ERROR] [1585063684.859991677]: An exception has been thrown: set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
[D435/realsense2_camera_manager-5] process has died [pid 3247, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/nvidia/.ros/log/07489960-6de4-11ea-b79c-00044bc48391/D435-realsense2_camera_manager-5.log].

I see these errors (and occasionally others) when launching through ROS and in realsense-viewer.  Generally unplugging and replugging the camera helps the issues to go away. Are there any workarounds for these issues to make performance with the camera more reliable?  Is there something wrong with my setup?  Are any updates to the debian packages for Jetson expected soon to be able to work with a newer version of realsense_ros?

Additionally, these issues seem more frequent when running at 30 fps versus lower.  Is the D435 excepted to be able to run without problems at 30 fps on a TX2?

Thank you for your help!

MartyG-RealSense, 2020-03-25:
The list of errors looks to me like an 'error cascade' caused by the first 'USB CAM overflow' error.  By cascade, I mean that once the first process fails then other processes that are dependent on it can fail.  Once the issue with the initial error is resolved then the subsequent listed errors can disappear.  This is why looking at the first error that appears in a log is a good general rule for debugging.

Regarding the USB overflow error: are you using the official short USB cable supplied with the camera please, or are you using your own choice of USB cable?

If you are using the official short cable with Librealsense, a way to increase USB stability on single-board computing devices such as Jetson is to use a mains electricity powered USB 3 hub instead of plugging the camera directly into the board.

cdimmig, 2020-03-25:
Thank you for your quick reply!  I actually only see one of the errors at a time.  The camera crashes either due to a USB CAM overflow or the IR steam start failure (and the following lines associated with that) or a few less common reasons. 

I am using my own choice of USB cable, but it is a brand new Anker USB C USB 3 cable.  I am currently using it with a Anker USB 3 USB hub. I can try a powered USB 3 hub.  Do you think that could be causing the other errors as well?

Are updates to the debian packages for the Jetson expected soon to be able to update from librealsense 2.28.1?

Thank you!

MartyG-RealSense, 2020-03-25:
Because RealSense cameras can transmit a large volume of data at higher FPS speeds / resolutions, it is important to have a high quality USB cable.  USB-C cables in particular can have quality issues.  So a good-quality cable suited to data transfer, rather than an inexpensive one for device charging, is recommendable.  Quality becomes especially important with cable lengths 2 meters and greater, where connections may degrade or fail unless the cable is high quality.

In regard to hubs, unpowered ones (known as 'passive' hubs) draw on the power supply of the computer that they are attached to (like devices that are directly connected to the USB port) and so are known as "passive" hubs.  Mains-powered hubs, because they are powered by the mains socket, are therefore independent from the computer's power supply and any instabilities it may have, and so are known as "active" hubs.

Mains-powered hubs are not a guaranteed cure for RealSense USB stability problems but do have a relatively high success rate at instantly resolving problems where a reset of the camera is required to correct the issue.

Regarding Debian package updates: I hope that the link below will be helpful.

https://github.com/IntelRealSense/realsense-ros/issues/1114

cdimmig, 2020-03-26:
I have assured that the cable I am using is high quality.  I will also test with a powered hub but have heard from others using a powered hub that the same issues occur just less frequently than I was seeing them. 

Similarly to the person in the linked issue, I only have version 2.28.1 available when I apt install. Where could I find the 2.32.1 versions as debian that was mentioned?

MartyG-RealSense, 2020-03-26:
Please look at the install instructions in the link below and try modifying them for 2.32.1 instead of 2.31.0.

https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785

cdimmig, 2020-03-26:
Thank you!  When I run `apt-cache showpkg librealsense2` (after apt updating) 2.28.1 is still the latest version that is available to me.  I'm running JetPack 3.3 to stay on Ubuntu 16.04 since that is what the realsense packages say they support, but in that post the person is running JetPack 4.3 (Ubuntu 18.04).  Is that necessary to get the latest librealsense packages and thus why I am not seeing 2.32.1?

MartyG-RealSense, 2020-03-26:
Someone else has had problems with the source for 2.32.1 on Jetson in the past week, even with the latest Jetpack updates.

https://github.com/IntelRealSense/librealsense/issues/6112

cdimmig, 2020-03-26:
Ideally I would like to install from the debian repository with apt, I'm just unclear if its necessary to have JetPack 4.3 in order to be able to apt install 2.32.1 or if there is some other issue.

MartyG-RealSense, 2020-03-26:
Going back to the specification you provided in your original message, I noticed that your ROS wrapper version is listed as 2.2.8 and your Librealsense version as 2.28.1.

A specific version of the ROS wrapper should ideally be matched with a specific version of Librealsense listed in the wrapper release notes.

https://github.com/IntelRealSense/realsense-ros/releases

Wrapper version 2.2.8 should be used with the older 2.25.0 version of Librealsense.  For the latest 2.33.1 SDK, ROS wrapper version **2.2.13** should be used.

cdimmig, 2020-03-26:
So since a version of the wrapper was not released for version 2.28.1 of librealsense (which is what is installed by apt on a TX2) and since it does not seem like there is a way to install version 2.33.1 on the TX2 (unless that has to do with the Jetpack version??), then is my only option to downgrade librealsense to 2.25.0 to work with wrapper version 2.2.8?

MartyG-RealSense, 2020-03-26:
They are questions that are outside of my experience unfortunately.  I will tag the wrapper developer @doronhi again.  Apologies for the delay in an answer.

doronhi, 2020-04-01:
checking on my Jetson-Nano now, librealsense2 version 2.33.1 is available as a debian.

cdimmig, 2020-04-01:
With what version of Jetpack / Ubuntu?

MartyG-RealSense, 2020-04-20:
@cdimmig   Do you still require support on this case please or can it be closed?   Thanks!

MartyG-RealSense, 2020-04-26:
Closed due to no further comments received.

closed 2020-04-26 by MartyG-RealSense