realsenseai/realsense-ros #1148: [D435] Spurious pointcloud behind object

SwiftGust, 2020-03-31:
I'm using Realsense D435

I found the pointcloud output is very spurious behind the cylindrical object in front (see below picture)
But I don't want any shadows behind actual obstacle distances
How can I get more accurate, not spurious results?

Resolution tested 
- 480 x 270 / 60 fps 
- 640 x 480 / 30 fps 
with using HighResMediumDensity preset parameters

5 cylindrical object in front and nothing else in the scene, pointcloud starts from the cylindrical object and very spurious behind the object. 
[image]

MartyG-RealSense, 2020-03-31:
The pattern of the black areas makes me think that it probably is not a problem  related to light reflecting off the surfaces of the cylinders, as the black areas would likely be more confined to the shape of the cylinders instead of spreading out across the image.

It may be worth observing the same scene with the RealSense Viewer program if you are able to and comparing the results to your own "Camera" program to see whether the code in your application has an influence on the results.  

Also, considering that you are using very low resolution settings, I wonder if "LowResMedDensityPreset.json" may be a more appropriate json to use, since the 'Medium Density' type of jsons are resolution-based.

SwiftGust, 2020-03-31:
Hi, 
I did additional test with all available parameter set provided.
I chose HighResMedDensity preset since it showed good quality regardless resolution while I was testing across some time ago.

As far as I tested, HighResHighAccuracy preset shows good results regardless of the resolution again, prevents this spurious depth & pointcloud issue.

Shortcoming is that cylinder object becomes too thin and black areas are quite large. 

Realsense Viewer
[image]

ROS Demo Pointcloud 
[image]

MartyG-RealSense, 2020-03-31:
I tried to recreate the results that you got with my D435i camera but couldn't.

Would it be possible please to perform a Dynamic Calibration of your camera to eliminate the possibility that the problems with the image result from camera mis-calibration?   The Dynamic Calibrator Tool can be found at the link below.

https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API

If you are using Linux, the installation instructions for Linux are on page 14 of the Calbrator's user guide:

https://www.intel.co.uk/content/www/uk/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

SwiftGust, 2020-04-01:
In my test environment, I tested both used D435 camera & new D435i camera and both exhibit similar problem.

During trying to work with dynamic calibration tool however, the tool gives me following error while other rs-* example & realsense-viewer programs does work fine in this environment.

None other realsense programs are not running at the same time, plug-out & back-in doesn't help on this matter.

Should I use librealsense built with libusb?

```bash
$ Intel.Realsense.DynamicCalibrator -v
2.6.8.0

$ Intel.Realsense.DynamicCalibrator -list
terminate called after throwing an instance of 'rs2::backend_error'
  what():  get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
Aborted (core dumped)

$ Intel.Realsense.DynamicCalibrator 
terminate called after throwing an instance of 'rs2::backend_error'
  what():  get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
Aborted (core dumped)

```

- OS : Linux Ubuntu 18.04
- Kernel : 5.3.0-42-generic #34~18.04.1-Ubuntu SMP Fri Feb 28 13:42:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
- Realsense : SDK v2.33.1
- D435 Firmware : 0.5.12.03.00
`librealsense2-dkms is already the newest version (1.3.13-0ubuntu1).`

P.S. In the realsense-viewer program I can see On-chip calibration & Tare Calibration and these are working, are these do exact same as dynamic calibration tool?

MartyG-RealSense, 2020-04-01:
The calibration options in the RealSense Viewer are quick and easy "lightweight" functions for depth data improvement that do not provide the full camera calibration that the Dynamic Calibrator tool does.

Do you get the same busy error please if you provide the full path:

/usr/bin/Intel.Realsense.DynamicCalibrator -list

If you are getting similar results with a new D435i though then it may not be a calibration related problem, as the cameras are calibrated in the factory and usually work well when new out of the box.

SwiftGust, 2020-04-01:
Result is the same with full path execution
```bash
$ /usr/bin/Intel.Realsense.DynamicCalibrator -list
terminate called after throwing an instance of 'rs2::backend_error'
  what():  get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
Aborted (core dumped)
```

MartyG-RealSense, 2020-04-01:
As you got similar results from a new D435i, let's put the calibration aside for a moment and look at other possibilities.  Do the cylinders have a reflective surface such as shiny metal, please?

SwiftGust, 2020-04-01:
Nope it's PVC or similar plastic pipe and quite a lot of dirts on it but light source would be at behind & up may from camera view-point.

Here is picture of the test environment.
[image]

MartyG-RealSense, 2020-04-01:
I don't think that the large black areas at the sides are shadows, because the shadows cast by the cylinders seem to already be accounted for in the image (the blue triangular shapes at the base of the cylinders on your point cloud image roughly correspond to the shadows on the RGB image).

During tests with the RealSense Viewer, I noticed that I seemed to get better alignment of the black areas around objects if I used Color as the texture source instead of Depth.  The process to do this is:

1.  Enable the RGB stream
2.  Enable stereo streaming in 3D point cloud mode.  
3.  At the top of the screen, use the drop-down menu to set the Texture Source to Color instead of Depth.

[image]

SwiftGust, 2020-04-01:
Thank you for the tip.
Matching depth with color stream seems helps pointcloud quality even for transparent as described in
https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_under_water.ipynb

But it doesn't look to me it does improves quality of depth over solid object
the downside seems much reduced FOV of pointclouds and 
possible additional processing on host side(?, not sure) 
Lots of side portion of image is thrown away to matching with RGB FOV from depth image
[image]

MartyG-RealSense, 2020-04-01:
If you enable the Infrared 1 stream instead of RGB, that should give you the option to use IR as the texture source instead of RGB.  Does that give the same restricted point cloud view, please?

SwiftGust, 2020-04-02:
Yes, setting "pointcloud_texture_stream" to "RS_STREAM_INFRARED" for realsense-ros does give the same FOV excluding black regions
and to see beyond using without color stream matched parameter "allow_no_texture_points" to "true"
as in README

Then the question is using pointcloud matched with texture does really gives improves quality of depth/pointcloud? 
In static tests, this doesn't seem improves pointcloud better and requires additional processing, seems just post-processing to me.

MartyG-RealSense, 2020-04-02:
Intel's camera tuning guide has the following information on use of IR:

*******************

The “LEFT IR” camera has the benefit of always:
1. Being pixel-perfect aligned, calibrated, and overlapped with the depth map, 
2. Perfectly time-synchronized, 
3. Requires no additional computational overhead to align color-to-depth, and
4. Gives no additional occlusion artifacts, because it is exactly co-aligned. 

The main draw-back is that it will normally show the projector pattern if the projector is turned on.

SwiftGust, 2020-04-06:
I don't think stream matching doesn't take additional processing based on what I tested.
I'm using D435 on embedded device and as I tested using left IR matching with 

- Depth stream : 640 x 480 / 30fps
- IR Stream 1(Left) : 640 x 480 / 30 fps

CPU usage on both cases with only changing pointcloud_texture_stream parameter

-  RS2_STREAM_INFRARED : CPU usage 120% ~ 160%
-  RS2_STREAM_ANY : CPU usage 30% ~ 70 %

CPU clockspeed is set to fixed in this case. 
Stream matching does apparently easier to distinguish from human eyes.
However I'm not sure using IR stream matching does give better quality pointcloud too.

MartyG-RealSense, 2020-04-06:
If you are using an embedded device with a limited hardware specification then there may be a limit on how much you can improve results, as you may have to use low resolution and not be able to use speed boosting methods such as offloading CPU processing work onto a GPU.

One way to improve the point cloud may be to use two cameras and place them so that their fields of view overlap.  In the areas where the views overlap, you may get better depth detail because there is redundancy in the data (both cameras are observing the same coordinates).  Having two cameras attached to the same device may not be practical though if the embedded device has a low hardware specification or a lack of additional USB ports.

May I ask please what your project goal is in scanning the five cylinders?

SwiftGust, 2020-04-07:
I did compiled realsense SDK with CUDA but GPU usage is quite minimal, only running D435 to output pointcloud is just using GPU up to 5%, is there more way to utilize GPU further?

I'm using D435 to generate obstacle map around my mobile robot, obstacle may appear in different shape, types including cylindrical object and so on.

MartyG-RealSense, 2020-04-07:
The RealSense SDK can offload work to GPU with the GLSL Processing Block.  Its effectiveness is limited on low-power devices though.

https://github.com/IntelRealSense/librealsense/pull/3654

I wonder if it would help to set a Depth Clamp so that the camera excludes depth data more than 1 meter from the lens.  This would give it enough time to do obstacle avoidance but prevent the wall behind the cylinders from being rendered.

A quick way to test with the RealSense Viewer whether this would work is to go to the 'Post Processing' section of the Viewer's side options panel, enable the 'Threshold Filter' and set its maximum to '1' meter.

[image]

SwiftGust, 2020-04-07:
I just look through rs-gl example and it seems most of realsense-gl functions for windows & rendering and realsense-ros code is quite different from rs-gl examples, couldn't find where the pointcloud is processed, calling process function to pointcloud filter does the job internally 
https://github.com/IntelRealSense/realsense-ros/blob/3ab5077763e980ae72c0c29aec4a697067d43744/realsense2_camera/src/base_realsense_node.cpp#L1480

can you point me how to utilize GLSL in realsense-ros?
just replacing rs::pointcloud to in the following line would offload cpu processing or does need more to utilize it?
https://github.com/IntelRealSense/realsense-ros/blob/3ab5077763e980ae72c0c29aec4a697067d43744/realsense2_camera/src/base_realsense_node.cpp#L1068
```cpp
_filters.push_back(NamedFilter("pointcloud", std::make_shared<rs2::gl::pointcloud>(_pointcloud_texture.first, _pointcloud_texture.second)));
```
I applied this and doesn't seems realsense-ros utilize GPU.

Plus, will realsense-ros support realsense-gl library build option later?

In realsense-ros, applying clip_distance parameter does this job.
Is this the same called threshold filter in realsense-viewer?

MartyG-RealSense, 2020-04-07:
I was not able to find an equivalent program to rs-gl for ROS to offload CPU work to GPU, unfortunately.  I will link the ROS wrapper developer @doronhi to this discussion for advice on this question of whether the ROS wrapper has GL support or has any plans to add it.

clip_distance is probably different from the threshold filter (there is more than one way to create depth clamps), but may be worth using if it achieves the same end result in ROS as a threshold filter.

forestbeach, 2020-04-09:
I get a weird image when I set up the camera.  Unusable right now

[image]
[image]

MartyG-RealSense, 2020-04-09:
Could you repeat the On-Chip Calibration in the RealSense Viewer please, pointing it at a flat wall, and see if that corrects your image?

forestbeach, 2020-04-09:
It was on 3D; 2D worked.  I have to check on 3D downloads.

MartyG-RealSense, 2020-04-26:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thank you.

MartyG-RealSense, 2020-05-01:
Closed due to no further comments.

closed 2020-05-01 by MartyG-RealSense