realsenseai/realsense-ros #1149: Not able to run D435i and T265 at the same time on RPI4

espentrydal, 2020-04-01:
Hi, I am trying to run the D435i and T265 at the same time on the Raspberry Pi 4 (plan to use the T265 to provide visual odometry for Ardupilot). I can start one node, but the second one will fail. Both if I start them separately in different terminals and if I start them together.

Thinking the difficulty could be linked to #1051 I've upgraded to the latest development version of both librealsense2 and realsense-ros to get the fix in #1088 

Info |  
-- | --
Camera Models | D435i  and T265
Firmware Versions | 05.12.03.00 and 0.2.0.926
Operating System & Version | Rasbian Buster
Kernel Version (Linux Only) | 4.19.83-emlid-v7l
Platform | Raspberry pi 4
SDK Version | 2.34.0
ROS verision | Melodic

`Device info:
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	841512070397
    Firmware Version              : 	05.12.03.00
    Recommended Firmware Version  : 	05.12.03.00
    Physical Port                 : 	2-1-7
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	903223053974
    Firmware Update Id            : 	903223053974

Device info:
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	845412111494
    Firmware Version              : 	0.2.0.926
    Physical Port                 : 	2-2-8
    Product Id                    : 	0B37
    Usb Type Descriptor           : 	3.1
    Product Line                  : 	T200`

First I start the D435i with these parameters to try to limit the bandwidth: `roslaunch realsense2_camera rs_camera.launch camera:=d435 serial_no:=... filters:=pointcloud enable_infra
1:=false enable_infra2:=false depth_width:=480 depth_height:=270 color_width:=424 color_height:=240 depth_fps:=15 color_fps:=15
 unite_imu_method:=false enable_color:=true enable_gyro:=true enable_accel:=true gyro_fps:=200 accel_fps:=63 initial_reset:=tru
e align_depth:=true`

I get a couple of warnings linked to lack of calibration and then at the end:
`01/04 10:07:39,633 WARNING [2701108240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data
available, number: 61`
It streams images ok.

In another terminal I then run `roslaunch realsense2_camera rs_t265.launch camera:=camera serial_no:=...`

I get:

`[ INFO] [1585732188.531716746]: RealSense ROS v2.2.12
[ INFO] [1585732188.531849298]: Running with LibRealSense v2.34.0
[ INFO] [1585732188.588640383]:
 01/04 10:09:48,617 ERROR [2769273872] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
terminate called after throwing an instance of 'rs2::error'
  what():  failed to set power state
[camera/realsense2_camera_manager-1] process has died [pid 2170, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/pi/.ros/log/322e9afc-73f8-11ea-ac23-dca63243a60b/camera-realsense2_camera_manager-1.log].
log file: /home/pi/.ros/log/322e9afc-73f8-11ea-ac23-dca63243a60b/camera-realsense2_camera_manager-1*.log`

In the other terminal I get

`ERROR [2646598672] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 01/04 10:19:39,932 ERROR [2646598672] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
[ERROR] [1585732167.649643240]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info`

camera-realsense2_camera-2-stdout.log:
`ESC[0m[ INFO] [1585732187.674162165]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory
to manager realsense2_camera_manager with the following remappings:ESC[0m
ESC[0m[ INFO] [1585732187.703471306]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] could not connect to host [navio.lan:58877], waiting...ESC[0m
ESC[0m[ INFO] [1585732187.772366334]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.ESC[0m
ESC[0m[ INFO] [1585732198.607606035]: Bond broken, exitingESC[0m`

camera-realsense2_camera_manager-1*.log does not exist. Let me know if you want me to publish any other logs.

Any way to fix this or is this just too much for the RPi4?

espentrydal, 2020-04-01:
@lrustand

Jenny0420, 2020-05-11:
**i have some error on ubuntu16.04 + realsense D435**
jonas@jonas-lenovo-ideapad-s410:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/jonas/.ros/log/36075f3a-9395-11ea-8675-28e34769b836/roslaunch-jonas-lenovo-ideapad-s410-16836.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jonas-lenovo-ideapad-s410:34735/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
[... 59 lines cut ...]
[ INFO] [1589207966.652497041]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589207966.652522477]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589207966.652537850]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589207966.652551852]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589207966.652578848]: num_filters: 0
[ INFO] [1589207966.652601112]: Setting Dynamic reconfig parameters.
[ INFO] [1589207966.714569169]: Done Setting Dynamic reconfig parameters.
[ INFO] [1589207966.725735090]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1589207966.726133188]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1589207966.726432596]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1589207966.746956159]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1589207966.747029812]: setupPublishers...
[ INFO] [1589207966.750498060]: Expected frequency for depth = 30.00000
[ INFO] [1589207966.789951170]: Expected frequency for infra1 = 30.00000
[ INFO] [1589207966.825508060]: Expected frequency for infra2 = 30.00000
[ INFO] [1589207966.862004847]: Expected frequency for color = 30.00000
[ INFO] [1589207966.904155520]: setupStreams...
[ INFO] [1589207966.905450339]: insert Depth to Stereo Module
[ INFO] [1589207966.905531736]: insert Color to RGB Camera
[ INFO] [1589207966.905595440]: insert Infrared to Stereo Module
[ INFO] [1589207966.905683457]: insert Infrared to Stereo Module
[ INFO] [1589207966.936453883]: SELECTED BASE:Depth, 0
[ INFO] [1589207966.954534546]: RealSense Node Is Up!
 11/05 22:39:27,158 ERROR [140028416096000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 11/05 22:39:27,359 ERROR [140028416096000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)

closed 2020-12-15 by RealSenseSupport