realsenseai/realsense-ros #1150: Multiple Camera Module Support

kyumanlee84, 2020-04-02:
Does the Intel ROS wrapper have to launch each node per D430 camera module? Or can multiple cameras be handled by one ROS node? If possible, how many cameras are handled by one ROS node? If so, does it publish each msg for each camera? In what order and how? If multiple cameras are processed in one node, is there any expected load?

MartyG-RealSense, 2020-04-02:
Further to my earlier answer on the Intel Support forum, I did more research on the subject of multiple cameras from a ROS node.  The documentation for the ROS wrapper describes how to define multiple camera serial numbers for a launch file called **rs_multiple_devices.launch**

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

The instructions do not specify an upper limit for the number of serial IDs that you can define for the launch instruction.  If the cameras are all attached to the same computer though then there may be a practical limit of the number of cameras that can be used simultaneously due to consumption of resources on the computing hardware.

On a powerful single-board computer such as Up Squared, for example, two cameras can be supported, whereas lower-spec boards such as the original Up Board and Raspberry Pi boards that are not RPi 4 may only be able to handle one camera .  On a full PC with a strong specification (e.g an i7 processor), it may be possible to use four cameras at the same time on the same computer.  

@doronhi the ROS wrapper's developer - who I have tagged into this discussion - may be able to answer how messages are handled for multiple cameras when using **rs_multiple_devices.launch**

doronhi, 2020-04-02:
Each node handles 1 camera. The number of nodes that the machine can handle vary, as @MartyG-RealSense states.

MartyG-RealSense, 2020-04-02:
@doronhi  Thanks so much for your help on this case!

MartyG-RealSense, 2020-04-15:
Case closed due to no further comments.

closed 2020-05-30 by MartyG-RealSense