realsenseai/realsense-ros #1152: Bug in set_cams_transforms.py

nikostsagk, 2020-04-06:
In the `set_cams_transforms.py` script, if you provide both values and `--file`, as in the [multicamera](https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines) example found in the wiki, it fails to unpack the variables.

Thus [line 77](https://github.com/IntelRealSense/realsense-ros/blob/3ab5077763e980ae72c0c29aec4a697067d43744/realsense2_camera/scripts/set_cams_transforms.py#L77) in the `set_cams_transforms.py` file should be changed into:
`x, y, z, yaw, pitch, roll = [float(arg) for arg in sys.argv[3:9]]`

MartyG-RealSense, 2020-04-06:
I will link this discussion to @doronhi the ROS wrapper developer.  Thanks for the report!

doronhi, 2020-04-07:
It seems you are right. I wonder how this example ever did work...
Would you care to push a PR?

nikostsagk, 2020-04-07:
Sure, I've just did.
Yeah, it's weird as I would swear that it worked for me as well..

MartyG-RealSense, 2020-04-26:
@nikostsagk   I will close this case now as you have pushed a PR.  The PR will remain open until the pull process is completed.  Thanks so much!

closed 2020-04-26 by MartyG-RealSense