realsenseai/realsense-ros #1153: Rviz doesn't display map

intel-user, 2020-04-07:
Hi every one, I'm trying SLAM with D435i camera following this tutorial [_https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i_] and I get this error.
when I run this command,
"**roslaunch realsense2_camera opensource_tracking.launch**"
rgbd.rviz appear but it doesn't display map and **Global Status: Error , Fixed Frame: Fixed frame [map] does not exist** , **No Image**
Terminal display this Warn
> [ WARN] [1586250842.291481430]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
> /rtabmap/rtabmap subscribed to (approx sync):
>    /rtabmap/odom,
>    /camera/color/image_raw,
>    /camera/aligned_depth_to_color/image_raw,
>    /camera/color/camera_info,
>    /rtabmap/odom_info

Sometime terminal display like this
> 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 
>  07/04 05:15:19,459 WARNING [140036175546112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
>  07/04 05:15:19,464 WARNING [140036200724224] (uvc-device.cpp:664) interrupt event received: 129, 186, 248, 203, 142, 193, 86, 30, 60, 97, 1, 0, 64, 97, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0

Can you help me fix that?
Thanks for your help.
[image]

MartyG-RealSense, 2020-04-07:
In the image that you have kindly supplied, you have a warning exclamation mark next to the 'Image' section.  Could you expand it open please so that we can see the settings inside the Image section please, like in the image from the tutorial shown below?  Thanks!

[image]

intel-user, 2020-04-08:
Hi @MartyG-RealSense , thanks for your reply
this's my capture, it doesn't have item **map**

[image]

by the way, I using ROS on Ubuntu16.04 on VirtualBox

MartyG-RealSense, 2020-04-08:
The troubleshooting documentation of the RealSense SDK recommends against using Oracle VirtualBox for a VM installation.  VMWare is recommend instead.

https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-im-using-a-virtual-machine-and-the-camera-is-not-working

https://github.com/IntelRealSense/librealsense/issues/2017

intel-user, 2020-04-11:
Hi @MartyG-RealSense , thanks for your reply :))
I tried install VMware and but it still not work.
Sometime I see this error like below image
[image]

Sometime it work at a few moment when start then it stuck
[image]
[image]

MartyG-RealSense, 2020-04-11:
The developer of the ROS wrapper has said "The warning message "interrupt event received" can be ignored and indeed will be removed in future versions".

I would place more significance on the "USB cam overflow" and "USB Rec Overflow" warnings.  This brings to mind a couple of points to check with you:

1.  Are you using the official short USB cable that was supplied with the camera or are you using your own, longer choice of USB cable?  A high quality cable is important for use with RealSense, especially with cable lengths longer than 2 m.  With an own-choice cable that has quality issues, if it is 2 meters or more long then the camera may experience degraded performance or disconnections.

2.  If you are using the official short cable and plugging it directly into the computer or into an unpowered USB hub (one that draws its power from the computer) then errors may occur if the stability of the supplied power to the USB ports is unstable.  A mains-powered USB hub can help to reduce the possibility of these errors occurring, since its power comes from the mains electricity socket and so it is independent of the computer's power supply.

intel-user, 2020-04-17:
Hi @MartyG-RealSense , thanks for your instruction.
I used the official short cable and plugging into  a mains-powered USB hub, but this error still occur
[image]
[image]
Are there something else u can advice for me? thank you

MartyG-RealSense, 2020-04-17:
Apologies for the delay in responding, as I was researching this issue with the 'Fixed frame [map] does not exist' error.  

Apparently, RViz needs to be able to find a robot's "transformation tree" (TF) and may give this error if it cannot find it.  An answer in the link below suggests that "map" is changed to "base_link".

https://stackoverflow.com/questions/52420672/ros-rviz-how-to-visualize-a-point-cloud-that-doesnt-have-a-fixed-frame-transfo

[image]

intel-user, 2020-04-20:
thanks for your instruction @MartyG-RealSense 
I use mains-powered USB hub and run `rs-fw-update -l` to check firmware and type of usb, it's usb2.1 and I change to another computer it's usb3.2. It's work fine
Thank you!

MartyG-RealSense, 2020-04-20:
I'm glad you found a solution - thanks for the update.   Good luck with your SLAM tests!   :)

closed 2020-04-22 by intel-user