realsenseai/realsense-ros #1156: How to get a large pointcloud using D435

sonumathur, 2020-04-08:
Hi 
I am new to ROS and i try to make a 3D Scan using Realsense D435.
i use following commands....

roscore

roslaunch realsense2_camera rs_rgbd.launch align_depth:=true

rosrun rviz rviz

and save the data using command : 

rosrun pcl_ros pointcloud_to_pcd input:=/sensor_msgs/PointCloud2

[image]

but i am not familiar with combine these pcd files to one.

what is methods to get a large point cloud or to comblne these files.....

sonumathur, 2020-04-08:
the command 
rosrun pcl_ros pointcloud_to_pcd input:=/sensor_msgs/PointCloud2
produce a large no. of files as in image.

MartyG-RealSense, 2020-04-09:
I hope that the link below will be helpful to you.

https://support.intelrealsense.com/hc/en-us/community/posts/360042935494/comments/360011165033

sonumathur, 2020-04-21:
@MartyG-RealSense ! Thanks.
I get a lot of help from the above given link.
Can provide help to combine .pcd or method to produce Large .pcd file.

MartyG-RealSense, 2020-04-21:
PCL has another tutorial for combining more than two PCD clouds using a method called **pairwise**.

http://pointclouds.org/documentation/tutorials/pairwise_incremental_registration.php#pairwise-incremental-registration

It offers a set of pre-made PCD clouds that can be used to test it.

https://github.com/PointCloudLibrary/data/tree/master/tutorials/pairwise

closed 2020-05-01 by MartyG-RealSense