realsenseai/realsense-ros #1157: RFE: SR305 xacro

henrychoi, 2020-04-08:
Hi, are you planning to support SR305 xacro?

MartyG-RealSense, 2020-04-08:
Thanks for posting your Feature Request!  I have labeled the message so that the RealSense team can track the request.  I will also link @doronhi the ROS wrapper developer into this discussion to seek his input.

MartyG-RealSense, 2020-07-08:
Hi @henrychoi   Intel carefully reviewed your request to add an SR305 xacro and decided that it was not a feature that is going to be supported.  Thanks so much for your patience.

closed 2020-07-08 by MartyG-RealSense

Tuebel, 2021-03-25:
Do you generally not want to support the feature or are you lacking workforce? We have internally put together our own description package for the SR305 and might be able contribute it to the upstream. So the question really is: Would you accept such a pull request?

MartyG-RealSense, 2021-03-25:
RealSense ROS users are very welcome to contribute pull requests for the ROS wrapper at the link below.

https://github.com/IntelRealSense/realsense-ros/pulls