realsenseai/realsense-ros #1158: catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release

Shumine, 2020-04-09:
Hello, I want to install real sense library for my ubuntu 18.04. 
but the error had occured when I type command below
"$ catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release"  at my terminal.
[image]

So I tried to do a lot of things to fix error..... 
And when I change  "-DCATKIN_ENABLE_TESTING=False" to
 " -DCATKIN_ENABLE_TESTING=True" , catkin_make command works successfully.

Now I have dont' idea what the "-DCATKIN_ENABLE_TESTING" thing(for configure catkin option??) works for "catkin_make".

Please let me know which file have this configuration code about this.  

Thanks

MartyG-RealSense, 2020-04-09:
Another RealSense ROS user had problems with this command in March 2020.  They found that the problem disappeared once they removed folders they were not using inside **joystick_drivers**.

https://github.com/IntelRealSense/realsense-ros/issues/1101

MartyG-RealSense, 2020-04-26:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thank you.

Shumine, 2020-04-27:
You can close this case now. 

I installed my ubuntu 18.04 again, and now I could install real sense library perfectly.

Thank you.

MartyG-RealSense, 2020-04-27:
Great news - thanks so much for the update!

closed 2020-04-27 by MartyG-RealSense